portapack-mayhem/mayhem-firmware #2392: Portapack H1 SV#8 at startup

gullradriel, 2024-11-24:
### Describe the bug.

I have an old H1 that is having a dead coin battery.
It was successfully using PMem to sdcard to save a few settings across power offs. 
I didn't updated it in long.
Now I didupdate it, and each time I'm starting it, I have a SV#8 at first power on, at the end of the startup icons animation.
Then,  I hit reset, and it's restarting and finishing without a bug.
I can't explain why it's not starting well at first boot.

### Reproduction

On a H1 with no coin battery and pmem to sdcard turned on:
- Turn off completely
- Turn on
- First boot is crashing on SV#8
- Hit reset
- Booting normally

### Expected behavior

Device is turning on and do not crash at first power on.

### Anything else?

- H1 boots correctly without pmem to sdcard
- Other devices having pmem to sdcard and no coin battery do not show the problem (H2M, H2+, H4M test unit)

zxkmm, 2024-11-24:
sounds still related to my splash change. will make a binary for you tot est

zxkmm, 2024-11-24:
https://discord.com/channels/719669764804444213/722101917135798312/1310268921420447874

zxkmm, 2024-11-24:
sorry that didn't noticed this:
> First boot is crashing on SV#8
Hit reset
Booting normally

then ya maybe it's not from me

gullradriel, 2024-11-24:
Yeah, no difference with your test bin. Same.

zxkmm, 2024-11-24:
glad to know that the issue is not from me.
I'll try reproduce your issue with my H1 too

gullradriel, 2024-12-01:
Tracked down the bug.
It strangely happens in 
```
bool should_use_sdcard_for_pmem() {
    return std::filesystem::file_exists(settings_dir / PMEM_FILEFLAG);
}
```

gullradriel, 2024-12-01:
Or not. Grrrr

gullradriel, 2024-12-20:
Fixed in latest stable (2.1.0)

closed 2024-12-20 by gullradriel