portapack-mayhem/mayhem-firmware #2398: APRS decoding

Waycco, 2024-11-29:
### Describe the bug.

APRS and AIS decoding not implemented well?

APRS works very poorly. Not even one APRS message has been received and I live in area where is a lot of APRS traffic. I can hear the APRS bursts but the APRS app can not decode it.

AIS works only if you are next to the ship. Normally AIS decoding should work from much longer distances.

Are these apps really tested enough?

ADS-B works well, Why APRS and AIS not?

### Reproduction

Test APRS and AIS receiving and
decoding your self do they work?

### Expected behavior

APRS and AIS signals should be received many miles distance.

### Environment/versions

At sea shore close to active vessel traffic

### Anything else?

Mayhem apps should be tested very well before they are put
to stable FW releases.

f1ghy, 2024-11-29:
I agree with that. I have noticed this too for a long time too.

gullradriel, 2024-11-29:
I don't know. Last time I got an AIS was from more than 20 km away.
You have to be in a clear of any obstacles environment or on the water yourself.
What antenna are you using ? What LNA, VGA, AMP ?

> Are these apps really tested enough?

Your tone is a bit rude, isn't it ?

Do you think we are paid or that we owe you anything ?

Please clone the repository, find a better way and push a PR, if you are able to :-)

f1ghy, 2024-11-29:
I wood like to compare BW... to the right one dble pressing DFU swich during "Audio app" but it move away too fast !
Ais show:
Rx BW: 1 750 000
Rx SampR: 2 457 600
Modulatn: 1
AM cfg: 0
NBFM cfg:same for APRS exept Rx SampR: 3 072 000

Waycco, 2024-11-29:
> I don't know. Last time I got an AIS was from more than 20 km away.
> 
> What antenna are you using ? What LNA, VGA, AMP ? 
> 
> > Are these apps really tested enough?
> 
> Your tone is a bit rude, isn't it ?
> 
> Do you think we are paid or that we owe you anything ?
> 
> Please clone the repository, find a better way and push a PR, if you are able to :-)

Sorry, maybe the tone might sound a bit rude but I think it is the fact that these mentioned "apps" don't work as well compared to ADS-B app for example. Just saying that it is not ideal to release "apps" with functional status if they don't work well. 

For example in many Youtube videos HackRF users talk about the AIS function but most of them say that I'm inside of the house or I don't live near the so I haven't tested the function. Maybe the APRS or AIS function is not that interested and that's why no one have noticed that they work quite poorly??

This works: https://www.wimo.com/en/picoaprs?srsltid=AfmBOopbBysz2gFZ_v1V_qjPxfW7nENm_YXndVcxXOCIl_ZnIa17sAA6

> What antenna are you using ? What LNA, VGA, AMP ?  Answer: normal whip antenna connected to H4M sma tuned to 161-162MHz..

I tested years ago some usb dvb-t dongle with his decoder and it worked much better compared to HackRF AIS app. 

Yes and nobody owe anything to anyone, but I just can't sit silent in home and hope that it starts to work by itself...

Software develops when there is coders, users, testers and people who find problems and report about them...

ToxxikHalo, 2024-11-29:
Apologies for jumping in like this but I feel it's justified...
I myself can neither code nor am I adept at even using the PP/HRF1 yet BUT one of the first things I learned and respected and are still in awes of is the incredible work done by a small group of rockstars in the community for the benefit of everyone! These people have full time jobs and full time families and lives of their own in between and STILL force 30 hours into a day to create something unique and awesome for strangers that too often don't appreciate the time and effort and frustration based into any one of the many projects going on in the mayhem community..... 
I don't believe you sir were intentionally being rude and don't mean to highjack this thread but the peeps who only benefit without contributing should respectfully take a step back to see the whole picture before question the quality or integrity of something gifted to the mayhem project by an extremely talented individual that usually goes unrecognized...  

Again this was more of a general pitch and not towards you Wayyco, plz don't take it as such...

I'm sure every one of the people who create the apps and programs are more than open to constructive criticism and opinion but you'll get nothing without giving them the deserved respect for what they've given already....

My apologies gullradriel, I lnow this isn't the place but I'm at work and had to be fast, lol... feel free to delete if it's inappropriate...

f1ghy, 2024-11-30:
Hello. 

 And verify that internal LNA amp give gain and not atténuation (it may burn). "Looking glass" app with waterfall is a quick an easy way to verify that.
 
Maybe, It must also be said that HackRF is a wonderfull general-purpose receiver, wideband, without filter stage,  and ONLY 8 BITS.
 Like everything that does a lot of things... it does it less well than a specialized device (which would not fit in our pocket).
But... It does quite well what we ask of it... (because it has a magnificent team, helpful, efficient, H24 because all around the globe.... makes it live, of course ;0) ) 

Thanks to them !
😘

Waycco, 2024-11-30:
Hello.

Ok, I got the point. Sorry about the too direct comments. Have to be more careful next time.

Waycco, 2024-11-30:
> Hello.
> 
> And verify that internal LNA amp give gain and not atténuation (it may burn). "Looking glass" app with waterfall is a quick an easy way to verify that.
> 
> Maybe, It must also be said that HackRF is a wonderfull general-purpose receiver, wideband, without filter stage, and ONLY 8 BITS. Like everything that does a lot of things... it does it less well than a specialized device (which would not fit in our pocket). But... It does quite well what we ask of it... (because it has a magnificent PortaPack team, helpful, efficient, H24 because all around the globe.... makes it live, of course ;0) )
> 
> Thanks to them ! 😘

Hi, 

- Yes LNA gives gain and it is working.
- My "whip antenna is a dipole" tuned to wave length using Nano VNA.
- LNA AMP set to 1 = ON

f1ghy, 2024-11-30:
Hi, I spend time to do test APRS for you.

I have nice results here, in the north of the France, with LNA on (1) and 32 and 40 (on an hackrf clifford edition).

Some stations may have a bandwidth larger than the 8,5khz. So you can try to increase your reception bandwidth in a file named rx_aprs.ini , in the setting folder. There, there is a line: nbfm_config_index. it should be 0 for  8.5kHz, 1 for 11kHz and 2 for 16kHz. you can try to change the 0 to 1.

Most station have 8.5Khz bandwidth, so nbfm_config_index=0 look to be fine.

I am not at home, and too far from any boat to make AIS test. There is also an rx_ais.ini

A way to modify these files is in "settings menu" then "app setting"

Make try and test, then tell us to participate.
Thanks.
;o)

Waycco, 2024-11-30:
> Hi, I spend time to do test APRS for you.
> 
> I have nice results here, in the north of the France, with LNA on (1) and 32 and 40 (on an hackrf clifford edition) and 1 24 32 on an regular r9.
> 
> Some stations may have a bandwidth larger than the 8,5khz. So you can try to increase your reception bandwidth in a file named rx_aprs.ini , in the setting folder. There, there is a line: nbfm_config_index. it should be 0 for 8.5kHz, 1 for 11kHz and 2 for 16kHz. you can try to change the 0 to 1.
> 
> Most station have 8.5Khz bandwidth, and a too large bandwidth may down the sensivity. But i have nice result with nbfm_config_index to 1, too. It maybe more tolerant in case of less tcxo/frequency stability but too large bandwidth may down the sensivity...
> 
> I am not at home, and too far from any boat to make AIS test. There is also an rx_ais.ini
> 
> A way to modify these files is in "settings menu" then "app setting"
> 
> Make try and test, then tell us to participate. Thanks. ;o)

Hi, thank you very much. I'll do that. Do you use full length antenna or 1/2 or 1/4?

Waycco

gullradriel, 2024-11-30:
@f1ghy thanks for the tests. Do you think adding a bandwidth control in the app is worth it, or should we leave it at the default ? It may be difficult to add as the GUI is already a bit full.  Maybe in a third tab, if it works.                                                                                                  

I will edit the APRS RX wiki page to reflect that it may need AMP 1 to have good results, and that you can tune the settings tuning if default is not enough.

gullradriel, 2024-11-30:
Edit the page to add the comments, in a "Important Notes" section:
https://github.com/portapack-mayhem/mayhem-firmware/wiki/APRS-RX#important-notes

f1ghy, 2024-12-01:
> Hi, thank you very much. I'll do that. Do you use full length antenna or 1/2 or 1/4?
> 
> Waycco

My advice is that a 1/4 or a dipole of 2x 1/4 work fine,  may be vertical or horizontal polarization (AIS, and APRS should be vertical polarisation). Dipole antena of 2 1/4 wave length may be done with télescopics antenas for diferents frequencies or with simples wires to put on the roof...
So it could be nice antenas easy to use, and easy to make yourself ;o)
(Others 1/2 or full lenght antenas need impedance matching or phase matching of diferents part)

f1ghy, 2024-12-01:
> @f1ghy thanks for the tests. Do you think adding a bandwidth control in the app is worth it, or should we leave it at the default ? It may be difficult to add as the GUI is already a bit full. [...]

It may be an idee... wait and see différents advices and test. An others way is to down the sample rate to make CPU work easyer if necessary (Remember there was questions about it for "pocsag app" some month ago, one year ago maybe).

f1ghy, 2024-12-01:
> My advice is that a 1/4 or a dipole of 2x 1/4 work fine,  may be vertical or horizontal polarization (AIS, and APRS should be vertical polarisation). Dipole antena of 2 1/4 wave length may be done [...]
You are welcome on discord mayhem too, if you need other or more information or help.
https://discord.gg/BjGyaKfB
;o)