portapack-mayhem/mayhem-firmware #2399: Can’t send signals

Niv1214, 2024-11-30:
### What happened?

.When I send signals, they are not received. For example, I try to open an electric gate and it does not open. It used to open, but now it does not. anyone know what’s the problem?

### Your hardware

Portapack h2

### Tests

- [X] I managed to update/boot previously

gullradriel, 2024-11-30:
Are you using the AMP ? Maybe it's toast ?
Can you show a video of what you are doing ?
Do you have another unit, or a flipper or equal that you can use to check the presence/absence of signal ?

gullradriel, 2024-11-30:
Maybe this ?

https://github.com/portapack-mayhem/mayhem-firmware/wiki/Preamplifier-IC-replacement

Niv1214, 2024-11-30:
Thanks I will check this. The antenna that I get with the Portapack broken
and I buy another one from AliExpress. Maybe the antenna is the problem?

בתאריך שבת, 30 בנוב׳ 2024 ב-23:48 מאת gullradriel ***@***.***

gullradriel, 2024-11-30:
It can be part of the problem. Not all antennas are suited to transmit.

Niv1214, 2024-11-30:
So I need 2.4ghz antenna right?

בתאריך שבת, 30 בנוב׳ 2024 ב-23:57 מאת gullradriel ***@***.***

gullradriel, 2024-11-30:
If you wanna transmit on that frequency, yes, and be sure to have one that is designed for both rx and tx

Niv1214, 2024-11-30:
Ok probably it’s the problem. Thank you very much for the help. If you have
discord or something we can talk and to do these things together, I've
recently become very interested in this business.

בתאריך יום א׳, 1 בדצמ׳ 2024 ב-0:33 מאת gullradriel ***@***.***

zxkmm, 2024-12-01:
@Niv1214 
Discord server exists: https://discord.gg/tuwVMv3

but if you are seeking something like “customer service” then no we don’t have it, since it’s a free project so developers have no duty to help all tickets in DM.  
It works more like: you post your thing in the discord server (or here) publicly, and if developers have time and willing to help, they’ll reply.

zxkmm, 2024-12-01:
Also, since you mentioned “it used to open, but now it does not”, although this description is lack of details, but if you only using it for example 10 times, and it’s not working further more, then you should investigate that maybe some of the gate/barrier control using rolling code, and it would only work a few times when your code is still in sliding window.