portapack-mayhem/mayhem-firmware #2400: Hackrf Portapack Meshtastic APP 915MHz Lora capability

Dollarstrawcanary, 2024-12-03:
### Description of the feature you're suggesting.

Add an app that will allows communication with specified meshtastic nodes/devices paired with bluetooth and transceive via 915MHz (US) Lora. Additional feature to decrypt 915MHz messages from unknow sources and make use of owned meshtastic mesh for FOX HUNT app telemetry. Use known and unknown local meshtastic nodes as repeters for increased 915MHz signal strength and decentralization of Hackrf signal origination.

Kopterlove, 2024-12-04:
Same as: #400, #693, #2171

gullradriel, 2024-12-04:
Closing duplicated issue.
Tracking is in https://github.com/portapack-mayhem/mayhem-firmware/issues/400

closed 2024-12-04 by gullradriel

Firecul, 2025-04-16:
It would be nice to have configurable frequency for other lora regions, EU uses 868 for example and others can use 433MHz.