portapack-mayhem/mayhem-firmware #2406: Stuck on the first screen

JvoixL, 2024-12-05:
### What happened?

I soldered the motherboard, installed portapack and then flashed the firmware for the first time. I tried 2.0.1 and 2.0.2, but they both stuck at the first screen and stopped. I followed the tutorial to restart and long-press the left and right keys, but it still couldn't start. The motherboard can be connected to the phone to use the RF ANALYZER APP, everything is ok.

### Your hardware

with DIY R10 mainboard and H2+ R4 portapack

### Tests

- [ ] I managed to update/boot previously

JvoixL, 2024-12-05:
So what could be the reason? thanks！

zxkmm, 2024-12-05:
the hackrf board looks weird for the "KX" label on it and has type-c. Maybe you should check of that.

WXLN, 2024-12-06:
买个成品hackrf吧，你这个故障很奇怪。

zxkmm, 2024-12-30:
This is hardware glitch. Since OP not providing further more info, I'm closing this ticket.

Please consider:

- Contact whoever sold you this, _asking them WHAT THEY HAD MODIFIED on this hardware revision_
- Refund this modified hardware.

closed 2024-12-30 by zxkmm

eried, 2025-01-03:
@JvoixL where you got this board? (btw: you should follow @zxkmm advice and contact the seller)

JvoixL, 2025-01-04:
> @JvoixL where you got this board? (btw: you should follow @zxkmm advice and contact the seller)

PCB from JLC PCB， I bought the original motherboard and soldered it myself.

JvoixL, 2025-01-04:
> @JvoixL where you got this board? (btw: you should follow @zxkmm advice and contact the seller)

I bought the electronic components and soldered the mainboard myself. PCB from Hangzhou Jiepei, China。

zxkmm, 2025-01-04:
Then, obviously it's either:
- something went wrong when you soldering it, for example: solder bridging, wrong components.
- this modified version of PCB requires different BOM and you use wrong one.

We can't help you because:
- it's not us who modified this PCB design.
- we can't check if you soldered everything correctly.

JvoixL, 2025-01-21:
thx for your reply，I‘ll check my soldered part.