portapack-mayhem/mayhem-firmware #2415: POCSAG: Show accented characters

Nikotine1, 2024-12-10:
### Description of the feature you're suggesting.

Hi,
I live in a region where I see a lot of French pocsag messages.
They use tons of accented characters and they all get replaced by a dot.

That's because of this piece of code in ``pocsag.cpp``
```// Translate non-printable chars. TODO: Leave CRLF?
if (ascii_char < 32 || ascii_char > 126)
    state.output += ".";
else
    state.output += ascii_char;
```

Is it really needed? Could we for example allow ascii characters up to 165?

[image]

Nikotine1, 2024-12-11:
I forked the code, removed the replacement of ascii characters > 126 by a dot, compiled and tested it.
It didn't work.
Words I know should have an é or à still show these letters as a dot, so that makes me think that perhaps they are encoded like that by the sender?

htotoo, 2024-12-11:
Pocsag uses 7 bit ascii, see: https://en.wikipedia.org/wiki/Radio-paging_code_No._1#Message_format
So nothing above 126 should be valid. Also <32 chars are control chars, so those doesn't make sense too.

Get more info about the receiver they uses, and maybe we can figure out, how they encode the message.

mfalkvidd, 2025-01-27:
Potential duplicate: https://github.com/portapack-mayhem/mayhem-firmware/issues/1406