portapack-mayhem/mayhem-firmware #2416: Put apps in bin file

Nikotine1, 2024-12-11:
### Description of the feature you're suggesting.

Now when you want to test some code, you need to move the compiled bin file to the device, and copy over the apps separately. This makes it difficult to switch back an forth between firmware versions, because each time you also need to switch the apps.

Could the apps be put inside the bin, so they are also changed at the same time?

htotoo, 2024-12-11:
Use the ppfw file,.and that has all in one. Flashable via web, and via internal flasher app.

gullradriel, 2024-12-11:
As stated there is already the ppfw format is exactly doing  that.

closed 2024-12-11 by gullradriel

Nikotine1, 2024-12-11:
Ah wonderful, I will update the wiki page then.