portapack-mayhem/mayhem-firmware #2417: Change on/off sequence

CRSourcing, 2024-12-12:
### Description of the feature you're suggesting.

Portapack accidentally switches on when in a bag or backpack and the encoder knob gets pressed due to movements or when putting something additional in the bag.  Is it possible to swap the sequence (double press to switch on, single press to switch off) to avoid this? Thank you.

gullradriel, 2024-12-12:
> Cut the battery cable and add a switch,

This. 

Some people also hackrf bank card format cards, cutting just the size of the knob tube, and insert it between the knob abd the board when they don't want the know to click.

gullradriel, 2024-12-12:
You can also try to see if your board is having the possibility mentioned in that old manufacturer documentation:
[image]
There will be no firmware side option, as the power on is a hardware thing.
Good luck !

closed 2024-12-12 by gullradriel