portapack-mayhem/mayhem-firmware #2418: Portapack H4M and HackRF One v2.0.2 wont boot. (Black Screen Only)

macalikm, 2024-12-12:
### What happened?

I have a new Portapack H4M and a New Great Scott Gadgets Hackrf One. The Hackrf One works as a stand alone. When I connect the HRF and the H4M, The LEDS on the HRF light, but the H4M Screen stays black. The unit will not connect to the Mayhem Hub Web Updater. The unit does not show in /dev/ttyACMx on a Ubuntu 22.04 host. The Mahem Hub Web App doesn't recognize a compatible device. I have tried everything in https://github.com/portapack-mayhem/mayhem-firmware/wiki/Won't-boot. I have performed Firmware and dfu updated on the HRF (v2.0.2). I have tried several different USB Data cables.

### Your hardware

H4M (Mayhem Edition)

### Tests

- [ ] I managed to update/boot previously

WXLN, 2024-12-14:
Perhaps you have purchased the most rare H4M without firmware flash

macalikm, 2024-12-14:
Hi WXLN,
Thanks for getting back to me.
I did manage to get it (mostly) working. 
I took an old HackRF One with a working Portapack H2 and swapped in the H4M, and it booted when I held the Left button. I was then able to update from the mayhem hub. So, it is apparently a HRF problem. 
I am still having a problem with persistent settings. the Date/Time resets when I power it down. I tried replacing the CR2023 Coin Cell battery, but no success there. 

I then Plugged the old Porta H2 onto the new HRF1 and it only starts in hackrf mode. I am not able to update that one.
https://hackrf.app doesnt see it connected as a serial device. I've tried it on Windows10 and Ubuntu Linus with a couple of known good USB Date Cables.

I cant get any help from GSG. Not sure where to go from here...

Thanks Again,
Mike

ToxxikHalo, 2024-12-14:
Have you tried this method???

https://rabbit-labs.com/how-to-flash-the-hackrf/

If you've done all this to a tee then check your drivers and make sure they're the correct ones.. and try installing them manually...
It's been a while since I've done this myself but I had similar issues and I did get it resolved.... lemme know how u make out... if I have to I'll go through the process again to remind myself and we'll go from there.... good luck

macalikm, 2024-12-14:
K, I looked it all over, and it looks like what I've already tried. I hadn't seen teh Rabbit's tutorial.
Ill go thru it step-by-step in the morning, and let you know how it goes. 
BTY, I just got the back-up battery working.
That turned out to be a defective coin batt holder.
Thanks again! Hopefully that'll fix it up

ToxxikHalo, 2024-12-14:
Yea that's fine git me tomorrow sometime and I'll get back to you asap and see what we can figure out... 
Yea the coin cell had to be that culprit in that case... 1 problem solved,  lol...
Aight ttyl bud

WXLN, 2024-12-14:
> 嗨，WXLN， 谢谢你的回复。 我确实设法让它（大部分）正常工作了。 我拿了一个旧的 HackRF One 和一个正常工作的 Portapack H2，换上了 H4M，当我按住左键时，它启动了。然后我就能够从 mayhem hub 进行更新。所以，这显然是 HRF 的问题。 我仍然遇到持久设置问题。当我关闭电源时，日期/时间会重置。我尝试更换 CR2023 纽扣电池，但没有成功。
> 
> 然后我将旧的 Porta H2 插入新的 HRF1，它只在 hackrf 模式下启动。我无法更新它。https: [//hackrf.app](https://hackrf.app)没有看到它作为串行设备连接。我已经在 Windows10 和 Ubuntu Linus 上尝试过，使用了几个已知良好的 USB 数据线。
> 
> 我无法从 GSG 获得任何帮助。不知道下一步该怎么办...
> 
> 再次感谢， 迈克

Insert the hackrf (or connected h2) data cable into the computer, first hold down the RST key, then hold down the DFU key and release the RST key. At this point, the driver should be recognized, and then use DFuxxx.bat to flash the firmware.

WXLN, 2024-12-14:
> > 你好，WXLN，谢谢你的回复。我确实设法修复（大部分）工作了。我拿了一个旧的 HackRF One 和一个正常工作的 Portapack H2，换上 H4M，当我按住左键时，它启动了。然后我就能够从混乱中心进行更新。所以，这显然是 HRF 的问题。我仍然遇到持久设置问题。当我关闭电源时，日期/时间会重置。我尝试更换 CR2023纽扣电池，但没有成功。
> > 然后我将旧的Porta H2插入新的HRF1，它只能在hackrf模式下启动。我无法更新它。https: [//hackrf.app](https://hackrf.app)没有看到它作为串行设备连接。我已经在Windows10和Ubuntu Linus上尝试过，使用了几条已知良好的USB数据线。
> > 我无法从 GSG 获得任何帮助。不知道下一步该怎么办...
> > 再次感谢， 迈克
> 
> 将hackrf（或者连接的h2）数据线插入电脑，先按住RST键，再按住DFU键，松开RST键。此时应该可以识别到驱动了，然后使用DFuxxx.bat刷入固件。

If your burning problem is still not solved through DFU, then you can consider that LPC4320FBD144 LQFP-144 is broken，However, this situation is rare,

macalikm, 2024-12-15:
I just went thru the steps again, I started with a new FAT32 32Gb SD card and installed the Copy to SD stuff and rebooted. 
Next, I tried https:// hackrf.app. 
Moved on to the bat files that come with the github firmware zip for v2.0.2

Followed the steps verbatim, but keep getting the same bad results.

What is LPC4320FBD144 LQFP-144?

++++++++++++++++++++++++++++++++++++++++
dfu_hackrf_one.bat failed with the following message: 
  DFU suffix CRC does not match
  A valid DFU suffix will be required in a future dfu-util release!!!
  No DFU capable USB device available

Then tried  portapack-h1_h2-mayhem.bin
Got this error message:
  Devices detected: 0 DFU, 0 HackRF
  ERROR: HackRF not found. 
  Please connect a HackRF One to your computer's USB port.
  If using a PortaPack, please put it in HackRF mode after connecting it.

  If your device never boot after flashing, please refer to https://github.com/portapack-mayhem/mayhem-firmware/wiki/Won't-boot

  Press any key to continue . . .
+++++++++++++++++++++++++++++++++++++++++

macalikm, 2024-12-15:
Hey  WXLN,
We got !

First, I used hackrf_spiflash to update the HRF firmware and boot loader (again).

Then:
I needed to go to mayhem_v2.0.2_FIRMWARE\driver and run  dpinst.exe
Then back to mayhem_v2.0.2_FIRMWARE and run flash_portapack_mayhem.bat
It booted up in portapack thee next time that I cycled power.

So, Now the Portapack H4M and the H2 are both working with older Genuine Great Scott HackRF Ones

Thanks again for all your help!

closed 2024-12-15 by macalikm