portapack-mayhem/mayhem-firmware #2419: NOAA receiving

NikCrYT, 2024-12-12:
### Description of the feature you're suggesting.

is it possible to implement NOAA satellite TX app or other ready to use app for receiving satellite images?
seen something similar in havocs modified firmware.

Smth like this:
[image]

gullradriel, 2024-12-12:
If someone works on it, maybe.
Anyway duplicate of https://github.com/portapack-mayhem/mayhem-firmware/issues/731 , please continue the discussions on that issue, thanks.

closed 2024-12-12 by gullradriel

Brumi-2021, 2025-03-25:
HI @NikCrYT , Why 433 Mhz ? 
I always decoded  NOAA Satellite APT using 137Mhz band.