portapack-mayhem/mayhem-firmware #2425: H4M Apps rx/tx not working / Apps freezing when exiting.

Brandon541, 2024-12-15:
### What happened?

#### First issue noticed— 
It doesn’t seem to matter which app, when using the back arrow at the top left corner of the screen the device freezes and has to be reset. 

#### Second issue noticed —
Whether it’s in ads-b, scope or search no signals are picked up by the device. I live really close to an airport and am seeing nothing. Additionally I have tried transmitting in the 2m band and am not seeing anything in the scope when 2m amateur radio is selected. I mean it’s just black. 

#### Steps taken —
1 turned it off and back on again
2 refreshed several different versions of mayhem incl. nightly releases.
3 reflashed dfu 
4 updated the Xilinx CPLD based on the guide.
5 attempted 4 different 32gb FAT32 MicroSD cards.
6 used the hackRF r10c device in hackRF mode with airspy and it works perfectly.

Other odd things that may be connected—
I have no issues flashing from windows or Linux. I get weird freezing when flashing from the web flasher.

microSD sometimes will show “fail: compare” if I try doing the test then proceeds to crash and show the meditation guru page.

[image]
[image]

### Your hardware

H4M and R10C

### Tests

- [X] I managed to update/boot previously

eried, 2024-12-16:
if it was not working from the beginning, I would contact the seller. Maybe the microsd slot has some soldering bridge or similar issue?

WXLN, 2024-12-16:
This question is quite strange
If there is no response from both transmission and reception, it may be due to damage to the U25 amplifier chip (hackrf one), and if the SD card is damaged, it may be a problem with the SD card. If there are issues with lagging and crashing, I suggest using your H4M with the power plugged in. If there are no problems with the power plugged in, then your battery is not working. Before doing these, I suggest you upgrade your hackrf one firmware to the latest version.

Brandon541, 2024-12-16:
> This question is quite strange
>
> If there is no response from both transmission and reception, it may be due to damage to the U25 amplifier chip (hackrf one), and if the SD card is damaged, it may be [...]

Much appreciated, don't think it is the u25 amp. If I run the hackRF in HackRF mode everything works as it should. It runs fine and gnu radio seems to be able to transmit just fine. I am on the latest stable firmware 2.0.2 and have tried the latest nightly build as well. I think it might be the SD card reader. At this point I'm gonna try and reflow the components.

Brandon541, 2024-12-16:
> if it was not working from the beginning, I would contact the seller. Maybe the microsd slot has some soldering bridge or similar issue?

I'm gonna test that theory and remove it tonight. Desoldering the sd card reader and resolder. Hoping this works.

closed 2024-12-18 by Brandon541