portapack-mayhem/mayhem-firmware #2441: Can't get into RF mode after recovering from DFU mode

Whiskaya, 2024-12-20:
### Describe the bug.

Tx, Rx and USB LEDs are dimmly lit
3V3 is fully lit
When connecting USB to Windows the device isn't recognized.
When entering DFU mode initially it wasn't recognized, then after waiting about 10 minutes disconnected, it was recognized as LPC
dfu_hackrf_one.bat worked and changed the device to "HackRF One" on device Manager, but got this:
```
Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2016 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

DFU suffix CRC does not match
A valid DFU suffix will be required in a future dfu-util release!!!
Opening DFU capable USB device...
ID 1fc9:000c
Run-time device DFU version 0100
Claiming USB DFU Interface...
Setting Alternate Setting #0 ...
Determining device status: state = dfuIDLE, status = 0
dfuIDLE, continuing
DFU mode device DFU version 0100
Device returned transfer size 2048
Copying data from PC to DFU device
Download        [=========================] 100%        43512 bytes
Download done.
unable to read DFU status after completion
can't detach
Resetting USB to switch back to runtime mode

Press any key to continue . . .
```
When running flash_portapack_mayhem.bat I get this:
```
Devices detected: 0 DFU, 0 HackRF
ERROR: HackRF not found.
Please connect a HackRF One to your computer's USB port.
If using a PortaPack, please put it in HackRF mode after connecting it.

If your device never boot after flashing, please refer to https://github.com/portapack-mayhem/mayhem-firmware/wiki/Won't-boot

Press any key to continue . . .
```
After that I tried everything on the link but using JTAG, since I just bought it and is being shipped.

**Did I fry my HackRF flash memory?**

### Reproduction

1. Go into DFU mode (press DFU and RESET; release RESET; release DFU button)
2. Run dfu_hackrf_one.bat 
3. Run flash_portapack_mayhem.bat (NOT WORKING)
4. Reset

### Expected behavior

Installs firmware

### Environment/versions

Windows 11 / Ubuntu / Raspbian / Mac

zxkmm, 2024-12-21:
Please try to flash on your Mac.

Whiskaya, 2024-12-21:
And just like magic, after several tries and configurations, the board accepted the flash from Zadig on Windows, as it was the only thing that could communicate to the HackRF.

This is what I tried:

- Remove the Portapack and flash WinUSB firmware from Zadig. Worked but didn't recognize the HackRF. When unplugging or resetting everything goes back and is only recognized when in DFU mode.

- Installed the Portapack and tried the same. Same result.

- Wait for 10 minutes with everything unplugged, then try to flash the HackRF without Portapack. WORKED, Windows recognized HackRF, but still didn't allow to run flash_portapack_mayhem.bat

- Used Zadig to flash WinUSB firmware to HackRF (it had v10.x and it downgraded to v6.x). Tried both "reinstall driver" and "downgrade driver". It got installed but still didn't allow to run flash_portapack_mayhem.bat

- Deleted the HackRF driver through Device Manager and installed every .exe in the drivers folder again.

- Installed the Portapack and tried everything again, and this time both bat files ran completely.

## Most probably these are a mix of flash memory in the device needing to power off completely and driver corruption at the computer.

Anyway, hope this helps the next guy!

closed 2024-12-21 by Whiskaya