portapack-mayhem/mayhem-firmware #2443: World map improvements

abrasius, 2024-12-24:
### Description of the feature you're suggesting.

A thought for apps using world map. As the map seems to be starting every time from lat 0.000000 long 0.000000, there should be a file where preset could be set. As the global map at low resolution is both horribly big to scroll and if resolution is increased to be practical, file size become stupendously massive, there should be ability to choose the local area map tile for the current location - maybe even switch to correct tile when scrolling the map around.

So, the requests:
* Central file to configure the location into for all apps using the map to read location preset from.
* App for user friendly way to create the location file; GPS module, user input, etc.
* Support for manageable sized map files with their coordinate grid defined either into a companion file or metadata in the map JPEG file itself

Benefits:
* location immediately correct when opening the app, less rather unnecessary wear on UI buttons
* smaller map size to handle, less load and memory usage

Thank you for your time, and highly likely improved and more practical implementation ideas regarding these thoughts.

htotoo, 2024-12-28:
What app opens map at 0,0 ?

abrasius, 2024-12-28:
Radiosonde receiver, for example. Alt/Lat/Lon all default to 0.

Waycco, 2025-01-19:
Maybe a support for GeoTIFF or some other known format in UTM coordinate system would be a working solution. Or perhaps a support for OpenStreetMap exported maps. Anyway if the coordinate system and the format of the map would be well known like GeoTIFF/UTM people could create their own raster map from anywhere they want with QGIS or some other GIS software.

eried, 2025-01-20:
The map implementations is very rudimentary, 

> * Central file to configure the location into for all apps using the map to read location preset from.
> * App for user friendly way to create the location file; GPS module, user input, etc.

This are easy to do, but normally the app should provide the position when opening the map

> * Support for manageable sized map files with their coordinate grid defined either into a companion file or metadata in the map JPEG file itself

This is more difficult, I am not sure how this will work with the limited resources there is.

htotoo, 2025-01-20:
I got this idea for map improvement: https://github.com/portapack-mayhem/mayhem-firmware/issues/2044

htotoo, 2025-09-09:
Some parts done in the OSM, other part of the issue is covered here: https://github.com/portapack-mayhem/mayhem-firmware/issues/2387

closed 2025-09-09 by htotoo