portapack-mayhem/mayhem-firmware #2446: FSK capture/replay support

igorolhovskiy, 2024-12-26:
### Description of the feature you're suggesting.

Would be nice to have an ability to capture/replay various FSK-type (2/4/GFSK, else?) modulated signals like https://xakcop.com/post/re-2.4ghz/. Nowdays many keyfobs are using these.

### Anything else?

Can support this part financially as not super good in C++ (and SDR in general).

zxkmm, 2024-12-28:
FSK is modulation, if you just want "capture and replay", you don't need to consider the modulation. 

If the signal itself is vulnerable to replay attack, and bandwidth is ok for HackRF, and doesn't have some frequency hopping and other tricky design, then it would just work with raw capture and replay.

zxkmm, 2024-12-28:
I just read the article that you mentioned. 

So it's a custom protocol and you probably need to write dedicated code for this specific remote.

Not sure if half duplex is good here tho.

igorolhovskiy, 2024-12-28:
"Capture and replay" is limited to 2M bandwidth. And FSK (as example of keyfob I have around) usually in a bigger range. So, I assume, it's needed some specific app for finding those frequencies and record them separately. No need for protocol decoding, just replay the same signal.
[image]

zxkmm, 2024-12-30:
Thank you for the clarification.

> And FSK (as example of keyfob I have around) usually in a bigger range

FSK is just a modulation, it's not the one that decides bw requirements. 
Your requirements need to write dedicated code, which is probably not possible for developers without the same device as you have.

So:
- We can leave this open and maybe someday a developer happened to have this device, then they can have a look.

On a side note, in your screenshot, to me, it seems only the pulses in the center (the strongest) one is real data. Other 4 looks like echo or ghost. I can't be sure with only a screenshot, but that's what it looks like at a first glance.

igorolhovskiy, 2024-12-30:
Thanks for your answer!

My idea is just record FSK modulation into file, means like 4 pieces of signal on a 4 different center frequencies, in this case no need to decode proto, correct?
Means instead of capture on one center frequency/bandwidth, you have some "droppings" means one sample on a Freq1/Bandwidth, second on a Freq2/Bandwidth and so on. Same on replay, I guess. 
I'm not sure it's possible to hop between frequencies range fast enough.

zxkmm, 2025-01-01:
It's physically not possible. 
data will be lost during hopping, unless the protocol of it meant to be hopping itself (for example BLE), and you probably can follow it by coding properly. 

and also as i mentioned above, in your screenshot, to me, it seems only the pulses in the center (the strongest) one is real data. Other 4 looks like echo or ghost. I can't be sure with only a screenshot, but that's what it looks like at a first glance.

igorolhovskiy, 2025-01-01:
Thanks for clarification!

zxkmm, 2025-01-01:
You are welcome. So I think we can leave this open until some developers got same device as you have and they can write it.