portapack-mayhem/mayhem-firmware #2451: OOK signal identifying, decoding, copying and transmitting.

ArsenAtomyan, 2024-12-29:
### Description of the feature you're suggesting.

Question with Enhancement

I need some help considering OOK.
The main ides is to copy my remotely activated garage door signal by decoding it and saving in SD card. I got regular remote with open/close/stop/block functions. I don't want to save the signal in IQ format cause it is quite (signal way) noisy in my neighborhood. So when I copy RAW signal from remote and then transmit it to everything works which tells me that my remote is using fixed code and not rolling(hopping) code.

So I use SubGhzD from Receive menu. It works and identifies my remote as

Type: Dooya 
Serial: 00530EE7
Data:
Bits: 40
Btn: 30
Cnt: 1
Data: 000000530EE7411E

But there is no [Copy Signal] button or [Save Signal] button like in Flipper Zero
So what can I do to decode and save this kind of signals?

Goal: By receiving OOK type signal copy and transmit it without using regular RAW/IQ file capturing.

Solution: Add buttons to SubGhzD utility for decoding, copying and transmitting those signals. (like in Flipper Zero)

Also Thank to all of you who are working on this firmware. I would be happy to join but I would be quite useless :P

DooMmen665, 2025-02-23:
The OOK application lacks a save button and the ability to enter the signal code in HEX form.

zxkmm, 2025-02-26:
> enter the signal code in HEX form

it's nearly not possible for human to calculate then enter OOk in HEX code unless you are Srinivasa Ramanujan

However you can use OOKEditor app to input code in binary