portapack-mayhem/mayhem-firmware #2454: Wake on RX

Miahporter, 2024-12-29:
### Description of the feature you're suggesting.

Would it be possible to have Mayhem wake from display sleep upon receiving and or decoding a received packet. ie in APRS app if the unit goes into display sleep mode once a signal is received with enough data to display received information the display would wake back up to show received info.

zxkmm, 2024-12-30:
yes it's possible, but probably need community to discuss would this request is widely needed among users.

Because one more switch and one more state control (either in p.mem or config file or in a var) need to be added to enable/disable this feature, and we don't have much rom/ram/p.mem spaces for some trivial thing.

Miahporter, 2024-12-30:
Gotcha. That makes sense. Either way thanks for your time

zxkmm, 2024-12-30:
You are welcome! Also if you wanna make yourself, it's fairly easy, you can check the implementation of stealth mode and turn off-screen things

htotoo, 2024-12-30:
It may not need a new switch.
Just an auto.screen off timeout pmem setting, and turning off the screen (or use the manual screen ott switch), and the wake on packet think. Maybe that should have a setting too.
(1 byte, where the highest bit is the boolmauto on on packet, the rest is a screen timeout value mapping)

zxkmm, 2024-12-30:
i mentioned switch is because maybe someone want packet rx to wake up, but some may not want, for example for those who would like to leave it open for few hours and back to check for fun.

htotoo, 2024-12-30:
I don't think, these should be changed so often, a simple setting page won't be enough. Switch would be an overkill

zxkmm, 2025-01-01:
that sounds acceptable to me :)