portapack-mayhem/mayhem-firmware #2462: Theme getting cyan with BLE RX application. Easter Egg? Bug?

Vanja678, 2024-12-31:
### Describe the bug.

There is an error when I go to **RECECIVE** -> **BLE RX** -> **CSV EXPORT** and type nothing in the **EXPORT CSV** field and just click **OK** click that the color theme is slightly different and some becomes cyan and some becomes black. And this only works if you have an SD card in your Portapack.

### Reproduction

1. Go to the **RECECIVE MENU**
2. Then open the **BLE RX** application.
3. Click on **CSV EXPORT**
4. Don't write anything in the textfield. Just click on **OK**
5. And then the theme is glitching.

> 1. [image]

> 2. [image]

> 3. [image]`

> 4. [image]

> 5. [image]

> 6. [image]

> 7. [image]

> 8. [image]

> 9. [image]

### Expected behavior

**I would have expected that there would be an error message saying: ***"CSV EXPORT ERROR"*****

gullradriel, 2024-12-31:
Bug. 
When we are doing it two consecutive times, hard failure.
It's because we didn't checked for empty text and are still trying to use it. It's messing up with memory.

Vanja678, 2024-12-31:
Ah okay thanks. That explains a lot to me.
**H A P P Y   N E W   Y E A R !**

gullradriel, 2025-01-05:
Fixed in #2468

You can try a test ppfw that I uploaded on discord [here](https://discord.com/channels/719669764804444213/722101917135798312/1325520014463336448).

closed 2025-01-07 by gullradriel