portapack-mayhem/mayhem-firmware #2464: Add Dutch C2000 to POCSAG APP

fvdijke, 2025-01-01:
### Description of the feature you're suggesting.

Is it possible to add the Dutch C2000 to the POCSAG app: 
Freq: 169.650 
Descr: P2000 
Norm: FLEX 
Baud: 1600
Modulaltion: 2FSK

Happy NewYear and Brgds

temmink, 2025-01-11:
I would like to have this feature as well, if possible.

gullradriel, 2025-01-12:
AFAIK:
Dutch P2000 is a public paging network in the Netherlands that operates on the POCSAG protocol, primarily used for emergency services like fire brigades, ambulances, and police. The system is part of the C2000 communications infrastructure, which is the national network for public safety and emergency services in the Netherlands.
It's using POCSAG protocol, and operate at 169.650 MHz. There is no particular data processing, so just changing the frequency in the POCSAG app should be enough.
You can already do it yourself, in the POCSAG app.

temmink, 2025-01-12:
> AFAIK: Dutch P2000 is a public paging network in the Netherlands that operates on the POCSAG protocol, primarily used for emergency services like fire brigades, ambulances, and police. The system is part of the C2000 communications infrastructure, which is the national network for public safety and emergency services in the Netherlands. It's using POCSAG protocol, and operate at 169.650 MHz. There is no particular data processing, so just changing the frequency in the POCSAG app should be enough ? Please tell me if I'm right or wrong.

Thanks for your reply and mentioning the frequency (e.g. 169.650 MHz). I can hear the typical sound (modem-sounds, verry clear)) on this channel, but the application is not showing the decoded messages.  However I see that when a signal comes in, it tries te decode them.

fvdijke, 2025-01-18:
> > AFAIK: Dutch P2000 is a public paging network in the Netherlands that operates on the POCSAG protocol, primarily used for emergency services like fire brigades, ambulances, and police. The system is part of the C2000 communications infrastructure, which is the national network for public safety and emergency services in the Netherlands. It's using POCSAG protocol, and operate at 169.650 MHz. There is no particular data processing, so just changing the frequency in the POCSAG app should be enough ? Please tell me if I'm right or wrong.
> 
> Thanks for your reply and mentioning the frequency (e.g. **169.650 MHz**). I can hear the typical sound (modem-sounds, verry clear) on this channel so the signal is there, but the application is not showing the decoded messages. However I see that when a signal comes in, (rx satu% 81) , it tries te decode them, i think.
> 
> (squelch set to 0, modulation type 1).
> 
> According to the Dutch P2000 messages, they are **1600 Baud** and **flex** encoding, instead of the normal 1200Baud and Modulation is **2FSK.**
> 
> In the Config section of the POCSAG app, i can save a log file. If needed I can post it here, because it isn't working according to me.

The current app only supports: Baud Rate: 05: 512bps, 12: 1200bps, 24: 2400bps (as far as i can see)
The P2000 system only works on 1600 baud.

temmink, 2025-01-18:
Indeed, I saw in the source code that it didn't support 1600 bps. Unfortunately I'm not qualified to change this meself. I tried to understand how the applications for the H4M work, and how to setup a development environment for this. But I'm not up to that task at this moment with my current knowledge ;-(

dark-juju, 2025-02-26:
So it looks like the change to add in the 1600 is relatively simple (update required in ~3 places) however the blocker for me making this change is:

1. I'm not familiar enough with the protocol to know if the flex encoding is something that the app already handles and therefore adding the additional frequency is enough. And, more importantly,
2. I am unable to test this change once made due to my geographic location not being in The Netherlands, and I feel a bit uncomfortable making changes I then can't test. 

If there's a way of getting confirmation on the first, then I do have a branch in a local fork lined up and could see if a PR is accepted into a nightly that someone else can test, but it does still leave a bad taste in my mouth not being able to test this myself.

fvdijke, 2025-02-27:
> So it looks like the change to add in the 1600 is relatively simple (update required in ~3 places) however the blocker for me making this change is:
>
> 1. I'm not familiar enough with the protocol to [...]

Thanks for your reply. If needed, i am more than happy to test. Unfortunately thats all i can do in this case.

dark-juju, 2025-03-03:
Hey fvdijke - Sorry for the slow reply, life stuff happened.

I have compiled the changes into a branch on my [forked repo](https://github.com/dark-juju/mayhem-firmware) and added a ["release" here.](https://github.com/dark-juju/mayhem-firmware/releases/tag/untagged-af2d96ab8b72fe6aeacc)

This is a massive long shot if this work as I do feel like I have just shoehorned in 1600 into a few files where this wasn't listed, and I have to assume there is more to it than that.

fvdijke, 2025-03-04:
Hi, thanks for the effort!
I loaded the new firmware (and apps) but no luck this time. The 1200Bd still works but the 1600Bd is not. It looks like there is no sync or handshake happening. It would be nice to have a debug option :)
If your capabilities ends here i completly understand.

Brgds.

Robbertjv, 2025-06-01:
Hi thanks for your amazing work, i enjoy the PP H4M.
I too second the desire for the P2000 add on to pocsag.

FYI

On the wiki referenced in the next link additional information on the protocol can be found.

https://www.sigidwiki.com/wiki/FLEX

It seems to be a implementation of motorola with a few more configurations. Samples are available on there too.

gullradriel, 2025-12-12:
Solved by https://github.com/portapack-mayhem/mayhem-firmware/pull/2883