portapack-mayhem/mayhem-firmware #2465: Last compilation 2.1.0 don't work with external apps

Unibit2025, 2025-01-01:
### Describe the bug.

Last compilation 2.1.0 https://github.com/portapack-mayhem/mayhem-firmware/releases/download/v2.1.0/mayhem_v2.1.0_COPY_TO_SDCARD.zip don't work with external apps

### Reproduction

 I don't have answer 

### Expected behavior

to work

zxkmm, 2025-01-01:
please define "don't work with external apps"

htotoo, 2025-01-02:
It is not enough to just copy the files to sd, you'll need to flash the fw too. (or not enough to flash, you'll need to copy to sd the new files).

If you done that, your sd may broken.

But some more details would help us to help you.

Unibit2025, 2025-01-02:
Thanks. After flash firmware all is Ok.

closed 2025-01-02 by htotoo