portapack-mayhem/mayhem-firmware #2467: Add ChinaRail "LBJ" message for POCSAG.

zjjxt, 2025-01-04:
### Description of the feature you're suggesting.

Add for receiving and showing POCSAG formatted ChinaRail "LBJ" message which has the meaning of train arrival alert. It can modified based on POCSAG. Here is the reference:https://github.com/killeder/GoRail_Pager

WXLN, 2025-01-23:
[有想法干翻老邵指日可待。实际也不是不能解决，[[

[image]

](url)
如我的破解方案

zjjxt, 2025-01-27:
> [有想法干翻老邵指日可待。实际也不是不能解决，[[
> 
> [image]
> 
> ](url) 如我的破解方案

能否分享下方法？或者提示下思路？

WXLN, 2025-01-30:
其实很好解决，只要搞明白了，为什么h2不能刷h3的固件，把这个问题解决，你就可以做到了。我简直是个天才，哈哈哈

zxkmm, 2025-01-30:
Hello, please try to discuss in English and if you have anything to share, feel free to submit a PR.
Thank you.

gullradriel, 2025-01-30:
Hello there.
Actually none of the open source contributors are working on it. 
@WXLN since you are a genius, I would happily review your pull requests :-)
Maybe you are having a good contact with the author of the application you put in screenshots. Please tell him to stop being shy, and to no fear: I'm reviewing the code, whoever the contributor is. 
Also: the list of contributors is based on the commits. Anyone who is having a pull request merged can have his name back in the contributors list.
Lastly, dear @WXLN , maybe you could try to appease things and make people go along instead of pouring oil on the fires. 
站着说话不腰疼 !

zjjxt, 2025-02-07:
> 其实很好解决，只要搞明白了，为什么h2不能刷h3的固件，把这个问题解决，你就可以做到了。我简直是个天才，哈哈哈

能透露下嘛？加个V？

zxkmm, 2025-02-07:
Please be aware that this is a place to talk about contributing open source code, and bug tracking, feature request and experience discussion.

Please go somewhere else instead of here to sell/buy your add-on PCB board that can bypass some close source Mayhem fork's paywall check.

This is a GPL open source free project, so close source paywalled fork isn't welcomed.

I will lock this convo to make it stick to it's original purpose: **A feature request about add decoder for ChinaRail "LBJ" message**, because off-topic talk already go too long and keep spamming project maintainers.

And if anyone have anything to contribute, either code/PCB/experience or anything, feel free to open a Pull Request. 

Thank you very much!