portapack-mayhem/mayhem-firmware #2469: Frequency Suggestion App?

jeremydbean, 2025-01-06:
### Description of the feature you're suggesting.

App that tells you if a particular frequency is generally AM, NFM, WFM, or if it can be multiple, which to select trying to capture a signal vs listen to voice, etc.  Wouod love it to include as well as the generally best settings for each type. Could be built on top of/replace the antenna length app. Just adding the extra data below.  

Would be amazing if you could tweak the gains here and get feedback, similar to how the amp and gains turn red/yellow in some apps when too high or low. Basically best “general” use settings for what most people would want to start with to use a particular frequency for whatever purpose, or what range of settings to use  for the start before tweaking on any given frequency.  

i.e. 

Frequency: 155.600
Usage: Voice
Type: NFM
Recommended: 16k
ANT500: (however many to extend, or if it’s out of range, recommend different antenna)

If possible maybe, once ideal settings are selected, some feedback in words not just waterfall. Basically a “yeah your settings are right but you aren’t getting good signal, check antenna type. Etc.). Even just as simple as: 

Signal Quality/RSSI: Poor/Average/Good/Great

mishazong, 2025-01-06:
Чому б не зробити  ковзну частоту , як у трансиверах , в полосі хоча б 200 кілогерц

htotoo, 2025-09-09:
This could be a standalone / ext app (ext app could use audio too for it)