portapack-mayhem/mayhem-firmware #2474: DroneFinder

Marto0o04, 2025-01-08:
### Description of the feature you're suggesting.

Receiving app that works on 5.8 GHz that scans one of  the A, B, E, F, R bands and switches to the next constantly and when it detects a signal the beeper can go off and analog video start displaying.

 (idk what I am talking about)

eried, 2025-01-09:
This is kinda your @gullradriel app no? but the bands are just fixed ranges

gullradriel, 2025-01-20:
@Marto0o04 :

For the detection and beep, you can use:

- Looking Glass (range detection)
- Level (but limited to one freq)

Maybe also Search, but not sure. None are starting a video live streaming; cause we do not have fpv video receive.

You would have better luck using a portable computer and the portapack in hackrf mode.

I'm sorry we would not have enough processing power for all at the same time.

closed 2025-01-20 by gullradriel