portapack-mayhem/mayhem-firmware #2476: NOAA Satellite Decoder

plastic-bottleneck, 2025-01-11:
### Description of the feature you're suggesting.

It would be nice to have a NOAA Satellite Decoder for the HackRF that can decode incoming data (136-138MHz) from the NOAA satellites as a **new functionality**. Being able to decode the incoming images from the satellites and saving them to the SD-Card, will be a great, useful and for some people; a teaching addition.
Important is, that the user is able to manually update the **TLE-Data** for the satellites over the SD-Card to calculate the next incoming satellite and doppler-shift.

### Anything else?

I do program and may try making it myself, but I don't know the programming structure from the HackRF as well as the main contributors.

gullradriel, 2025-01-12:
Duplicate from https://github.com/portapack-mayhem/mayhem-firmware/issues/731

To contribute, check the wiki, visit the discord.

We do not bite.

closed 2025-01-12 by gullradriel

Brumi-2021, 2025-01-12:
Hi, it is already  in our roadmap , but so far @htotoo and myself are trying to implement firdy maritime HF  Weather fax maps decoding 
. 
But, anyway ,  as you may know , you will need an additional good antenna + and probably and external LNA for that band . 

Cheers