portapack-mayhem/mayhem-firmware #2477: sliding frequency

mishazong, 2025-01-12:
### Description of the feature you're suggesting.

Hello. Why not make a sliding frequency in the band of at least 200 kilohertz, like in SDR transceivers?https://youtu.be/5tIoq_UqHy4?si=VgpB2264FS6r4_F6

gullradriel, 2025-01-12:
The guy in your video did some modifications, posted a bin, but never provided his source nor tried to make a pull request here.
Find the code, make a Pull Request ?

mishazong, 2025-01-12:
Therefore, I suggest you do it like in the video and post it here for all users. If you can do it that would be great. Thank you

gullradriel, 2025-01-20:
If you want it, ask the author of the video to provide his sources, or make a PR yourself.

You can already use a slider in Audio. The window is not as big as in the video, and you have to press OK to set the new frequency, but it's working.

mishazong, 2025-01-20:
[portapack-h1_h2-mayhem.zip](https://github.com/user-attachments/files/18481247/portapack-h1_h2-mayhem.zip) here is the slip rate file. I installed it and got a black screen. I restored via DFU. I can't figure it out. Maybe you can help. Thanks

gullradriel, 2025-01-20:
This is a compiled bin, and not sources. It's of nearly no use except flashing.

mishazong, 2025-01-21:
Too bad this isn't a source. I thought you could help me with this.

Viarety, 2025-02-04:
And what? Do we now have mayhem developers who can only interact with someone else's source code? The guy provided a video in which the sliding frequency function is 100% described and shown, and it seems to me that it is not a problem for a person who knows C++ and his business to add a function of this kind. Moreover, in the comments, the author says that he approached the developers of mayhem with the addition of this feature, but then this idea was rejected.

RocketGod-git, 2025-02-04:
@Viarety this is an open source project. Nothing is added without source, review, etc. It could be doing something nefarious in the background. 

Being a dick to people who can do things - to make them do the things you can't, seems like a bad way to go about this. 

Maybe I'll do it on my spare time, or if you toss me $300k USD I might be motivated to get it to you by next week.

Viarety, 2025-02-04:
This is your voluntary decision to make the project open source and free. It is your choice not to take money for use, not to take donations. But if you've already done that, then be kind enough to show at least some initiative. I read the problem about the color of the waterfall, raised a couple of years ago, when everyone asked for a color change and you had both the source code of the new color and the opportunity to simply choose a theme, but you corresponded with the author for several days until you got to the bottom of it completely and he said that fuck this idea and it would be easier for him to use if He likes it better. I'm not asking you to implement some impossible functionality, and in my opinion, even though I don't write in c++, this implementation is very simple. That's how you talk about motivation, well, then you don't have to do anything at all, abandon the project and do nothing if you don't have motivation. It's your conscious choice to do the project selflessly.

gullradriel, 2025-02-04:
@Viarety :

> And what? Do we now have mayhem developers who can only interact with someone else's source code? 

And what ? People can't choose what they are willing to work on for free ?

> The guy provided a video in which the sliding frequency function is 100% described and shown, and it seems to me that it is not a problem for a person who knows C++ and his business to add a function of this kind.

Then what are you waiting for to send your Pull Request, if it's so easy ? Maybe you'll have to start by learning C++, but it should be no problem, heh, since C++ is soooooo easy ?

> Moreover, in the comments, the author says that he approached the developers of mayhem with the addition of this feature, but then this idea was rejected.

> But if you've already done that, then be kind enough to show at least some initiative. I read the problem about the color of the waterfall, raised a couple of years ago, when everyone asked for a [...]

You are talking about this: https://github.com/portapack-mayhem/mayhem-firmware/pull/745. 

The user, [strijar](https://github.com/strijar) , had some pull requests merged, some were reverted because of user complaining too much after the changes.

Then, if you are seeing the PRs comments, at some point @strijar just had enough and closed his pull requests. 

We acted based on user feedback, here and on discord. You can see @strijar 's various pull requests [here](https://github.com/portapack-mayhem/mayhem-firmware/pulls?q=is%3Apr+author%3Astrijar) and as you can see, some were merged, some were closed, and some were reverted.

The color scheme thing was abandoned when it was asked to give the users a choice and make the color scheme configurable. We did not close his pull requests, he did. 

> This is your voluntary decision to make the project open source and free. It is your choice not to take money for use, not to take donations. 

We, as contributor, decided to contribute to a free, open source project. You can do the same. Let's see if you're gonna work for free as much as we did.

> I'm not asking you to implement some impossible functionality, and in my opinion, even though I don't write in c++, this implementation is very simple. 

How can you know, you don't f*cking write C++, you're just a troll wasting our time, ain't you ?

> That's how you talk about motivation, well, then you don't have to do anything at all, abandon the project and do nothing if you don't have motivation. It's your conscious choice to do the project selflessly.

Having motivation for something I want to do and having motivation for something someone else wants me to do are two different things. Yet we are all taking some of our time to contribute on things that are sometimes not interesting us, and even to answer users bad mouthing and using an insulting tone with us. Please understand that no one here is your f*cking dog.

# What you could have done instead of badly complaining 
- being polite
- read the wiki
- search for the code in the pull requests, mention the links instead of being blurry about it and make a shitstorm 
- propose modifications, try them yourself, learn to make a proper contribution
- gently discuss the issues like any decent human being should be

Viarety, 2025-02-04:
Ok

Viarety, 2025-02-04:
Still, it's your choice to work on this project for free, I wouldn't, so that's your craft, and you do it.

gullradriel, 2025-02-04:
@mishazong as you can see a positive effect of @Viarety 's shitstorm is that I used the search function and found the proposed modifications in some closed pull requests.

Viarety, 2025-02-04:
I'm not asking you to do anything, I'm just asking you to explain why you can implement such a simple function using only the source code.

gullradriel, 2025-02-04:
> Still, it's your choice to work on this project for free, I wouldn't, so that's your craft, and you do it.

Not only mine. It's all the whole contributors crafting, and user feedback.

Viarety, 2025-02-04:
Yes, I may have overreacted a little because I do not know c++, but I have a lot of experience with other programming languages and do not really understand why this is not so simple and the source code is extremely needed.

gullradriel, 2025-02-04:
> I'm not asking you to do anything, I'm just asking you to explain why you can implement such a simple function using only the source code.

If you read the whole story, you will see that in the PR there were a few problems that were never solved up to this point:
- audio app was hanging on after a while, reason unknown
- the spectrum color change modification were fought by users, and at that time the maintainers decided to stop investing time in it. When we discussed it, making it a choice was harder than just a few lines of C++, and it couldn't be externalized as a sdcard file.

I remember taking part in the discussion, and it's still on display if you're reading the related issues/Pull Requests. I'm not the one who closed or complained. I even stated that I liked both the ideas and colors, and encouraged the author to comply and finalize the things.

But you know what, maybe the author had enough, or it was boring for him, or he changed radios, or whatever ?

What  people can do now all the links are in the actual issue is:
- try to isolate sliding frequency code 
- compile it with actual firmware
- test and debug it
- propose a new PR

Maybe someone will do it, maybe not.

Viarety, 2025-02-04:
I'm not talking about color, I'm talking about a sliding frequency.

gullradriel, 2025-02-04:
> I'm not talking about color, I'm talking about a sliding frequency.

- audio app was hanging on after a while, reason unknown

What people can do now all the links are in the actual issue is:

- try to isolate sliding frequency code
- compile it with actual firmware
- test and debug it
- propose a new PR

Viarety, 2025-02-04:
Okay, thanks for the clarification, no complaints.

Could you take a look at my issue as well?

gullradriel, 2025-02-04:
> Guys, don't waste your time arguing. I'm also interested in this topic, otherwise I wouldn't have raised this issue. I've had Hack Rf for a while now and I'd like to improve it a bit. I think you can help me a bit with this. The topic is interesting. Thanks for your understanding .Sorry for the translation, this is Google Translate, I'm from Ukraine

No problem. As said, we know have a reference of the closed pull request which originally implemented the sliding thing ( https://github.com/portapack-mayhem/mayhem-firmware/pull/745 )

What's needed is:

- extract sliding code 
- add it back in a fresh next copy
- test and see if it's still hanging on after a while, fix it
- propose a new pull request with the changes

Thanks.

Viarety, 2025-02-04:
I can do these in the near future.

strijar, 2025-04-08:
I updated my Portapack (bought H4M) and want to get back into firmware development. I can take on this task.

gullradriel, 2025-04-08:
> I updated my Portapack (bought H4M) and want to get back into firmware development. I can take on this task.

I'll review with pleasure. Please keep it at one feature by PR's, and don't be afraid if we demand some bits of testing or adjustments ;-)

mishazong, 2026-09-09:
yes i tried . sliding frequency works well. thanks 👍