portapack-mayhem/mayhem-firmware #2481: "Empty" firmware.bin (with flash utility)

Waycco, 2025-01-18:
### Description of the feature you're suggesting.

Request for future development...

It would be nice to have an option that you could easily "empty" the whole firmware from HackRF/Portapack combo. Or you could have an option to update such firmware to the device that has no applications/function at all. 

Idea is that it makes the device just a device that has no functions, only the capability to update firmware from sd card (Only flash utility). So if you want to turn the device to a "non functional circuit board" you could upload this "empty firmware" and the device would be just a dummy computer with only one capability to receive/install new firmware from sd-card.

And when you want to make the device a fully functional again you could just load the latest package to sd card and run the installation of the official firmware with flash utility.

Also a good instruction to Wiki how to erase flash totally and install firmware.bin from scratch would be nice. 
And/or perhaps a tool for doing that could be another option...

zxkmm, 2025-01-18:
Hello, we already have it. Please check https://github.com/portapack-mayhem/shipping-firmware

Thank you.

Waycco, 2025-01-18:
Cool, didn't know that, thanks.

gullradriel, 2025-01-18:
> Also a good instruction to Wiki how to erase flash totally and install firmware.bin from scratch would be nice.
> And/or perhaps a tool for doing that could be another option...

https://github.com/portapack-mayhem/mayhem-firmware/wiki/Update-firmware

Waycco, 2025-01-18:
> Hello, we already have it. Please check https://github.com/portapack-mayhem/shipping-firmware
> 
> Thank you.

BTW, does the shipping-firmware support ExFAT on the SD?

gullradriel, 2025-01-18:
Hmm I don't think it was updated to support it.

Waycco, 2025-01-19:
I was just thinking that is Hackrfone/Portapack capable of booting via sd card slot or from the usb port? I mean could it be possible to create a bootable sd card or a usb stick that would install the latest firmware from the sd card or via usb port installed usb stick even in DFU mode?

zxkmm, 2025-01-20:
I think it’s impossible to do that, because we can’t do modification to DFU, because DFU is hard defined in the NXP mcu chip and no one can touch it.

But if you just want sort of “automatically” flash firmware then yes, it’s easy to modify the firmware to detect specific file name and flash on boot.  It however sounds meaningless because it cannot behave like sort of “recover device from bad situation” because if boot failed, then it cannot automatically flash, and if the boot succeed, then it’s pointless to flash again.

—————

And for your first question, if I read it correctly, you are asking if it’s possible to directly boot from sdcard, (like PlutoSDR I assume).
as far as I know, it’s not possible, because on this device, the sdcard support were provided by outer part (PortaPack and chibios) instead of the mcu itself. but those other SDRs that can boot from sdcard directly, has SDIO directly from the chip. (For Pluto, the ZYNQ has SDIO controllers in its PS)

I could be wrong about this tho, because I had never took a deep look of the NXP mcu on HackRF (and even if I take a deep look, I may not able to understand it completely), so I’m re-opening this issue so people who’s interested can check.

Waycco, 2025-01-20:
Hmm...does
this mean that if you put the H4M to DFU mode you would in all cases need a "smart USB stick" that has some host capabilities, so that it could detect the hackrf and push the dfu-util -a 0 -D firmware.bin command? So basically without an external computer it is not possible to boot and install the firmaware if the device have had a "blackout" and can be controled only via DFU?

zxkmm, 2025-01-20:
From what i understood, it’s not really possible to touch the DFU program, because that’s why it can be used as DFU. If you can modify it, it could be broken. 
Would you mind me asking why you want that such an edge case functionalities? Is it hard for you to have access to a computer?

gullradriel, 2025-01-20:
> Hmm...does this mean that if you put the H4M to DFU mode you would in all cases need a "smart USB stick" that has some host capabilities, so that it could detect the hackrf and push the dfu-util -a [...]

You need a computer to download the firmware anyway.

Waycco, 2025-01-20:
Mainly the reason is that if you need to use it in place where you don't have access to computer (or very poor access) and the amount of stuff you can carry is limited. And you need to travel with only shipping firmware installed. So to have self booting SD or USB could have been convenient in case something goes wrong when switching from shipping firware to the official firmware.

And also downloading and writng a bootable card in general might be a bit easier than uploading firmware with terminal commands. At least for basic users like me :)

I was also thinking that it would be nice if Hackrf.app could have an option to upload firmware and all the required commands when the device is in DFU mode? At least I tried it, but it was not able to detect the device in DFU mode. I understood that it could be at least theoretically possible via Google Web Serial API?

zxkmm, 2025-01-20:
DFU doesn’t have serial support. And even we completely implemented the DFU tool inside of MayhemHub with WebUSB instead of Web Serial or sort, we still cannot modify the DFU program of hackrf like I mentioned above.

zxkmm, 2025-01-20:
If what you actually want is just hide some apps in specific case, just use App Manager to hide apps

Waycco, 2025-01-20:
Oh yes, hmm..... I understand what u mean.....But still I think I'd prefer a "totally empty" shipping firmware with only tetris and pacman available :)

zxkmm, 2025-01-20:
You are welcome. 
However unfortunately regarding your request, since it’s very very edged case, and we don’t have much people here, so we don’t have energy to make one more fork and maintenance it. Because shipping firmware and app black list already completely did the best of the case you are thinking.
you can either do it yourself (it’s easy, just ask ChatGPT), or wait some developers to that, but I doubt if they’ll do that too.

closed 2025-02-14 by zxkmm