portapack-mayhem/mayhem-firmware #2483: Serial terminal / monitor application

Waycco, 2025-01-19:
### Description of the feature you're suggesting.

For future develoment. 

It would be nice if 
there was a small basic terminal serial
monitor that could listen, display and interact with some other device which are connected to the usb port. 

Like a simple Arduino IDE type serial monitor with touchscreen keyboard. This allows you to write for example AT commands and read/write data with other devices attached to hackrf/portapack.

zxkmm, 2025-01-20:
if you meant as master device then it's probably need really much work and im not capable to do that (also not sure if it's possible).
if you meant as slave device, then i'l writing it already (to help debug mayhem-cli app)

Waycco, 2025-01-20:
The idea was that you could connect a microcontroller (arduino or esp) to hackrf and then give text commands to the microcontroller using portapack keyboard and lcd display and also receive serial text from the microcontroller...

Waycco, 2025-01-20:
Yes that's better.

htotoo, 2025-02-07:
With USB only it is unlikely, but with the H4's gpio port with a microcontroller, it is possible. (hardest part is the keyboard implementation i think for the standalone app)