portapack-mayhem/mayhem-firmware #2484: Shutdown functionality

yath, 2025-01-19:
### Description of the feature you're suggesting.

Hi,

I got a Portapack H4M and plugging in a USB cable just for charging turns it on, no matter the switch position. I can turn the device into Sleep Mode while charging, but this leaves the RF on. Having a menu option to halt the processor(s) would be nice.

Thanks,\
Sebastian

gullradriel, 2025-01-19:
The RF is not on because the unit is on.
It's only on if you start a Radio RX or TX app, and only during RX or TX

gullradriel, 2025-01-19:
Put it to sleep or just set auto sleep in the settings.
AFAIK, we can not make a 'processor' shutdown.

gullradriel, 2025-01-19:
Last, if you do not push up the switch button, you are not charging the battery.

gullradriel, 2025-01-19:
When on the main menu you are at 4% on M0 and 0% on M4.
Pretty sure the 4% on M0 is the DFU menu I'm using to display the CPU consumption.
So... Little to no gain.

closed 2025-01-20 by gullradriel