portapack-mayhem/mayhem-firmware #2485: Bricked H4M Portpack Mayhem

roaderunner999, 2025-01-19:
### What happened?

New unit yesterday. Flashed the latest firmware and she ran beautiful all night.

Today, with battery full, it went black screen while in an app, and now when booting it only flashes white and goes to black.

When powered on or connecting a USB cable there is only a green power light and a faint red Transmit light

Wanted to try reflashing, but can not get any computers to see it. Have tried 5 cables, 2 computers, windows and ubuntu. I can set the unit to HACKRF or DFU mode but it can not be seen by a computer.

Tried to follow instructions for full reset
https://github.com/portapack-mayhem/mayhem-firmware/wiki/Full-reset

### Your hardware

H4M Portapack Mayhem

### Tests

- [ ] I managed to update/boot previously

gullradriel, 2025-01-19:
Looks like out of battery unit.
Do you know that the switch on the right have to be on / up for it to charge ?

ToxxikHalo, 2025-01-19:
Make sure you have the correct drivers on your computer... that's the issue w flashing a lot of the time... u might have to install them manually... I dealt with that once too.

Waycco, 2025-01-19:
If you put the the device to DFU mode like this:

1. Put the H4M battery switch off (down)

2. press the DFU button (keep down), then put usb cable to computer, if the computer asks (not in all cases) do you want to allow this device to be attached to computer or something like this then select yes, then release the DFU button.

3. When the device is in DFU mode, write this command to the bash (Linux/Mac terminal): dfu-util --list

You should see the device now?

roaderunner999, 2025-01-19:
thanks guys, i just opened it up and reseated the two boards and now it's fine.

closed 2025-01-20 by eried