portapack-mayhem/mayhem-firmware #2487: Freqcy Scanner/Spectrum analyzer with CSV Output

SGK-IoT, 2025-01-21:
### Description of the feature you're suggesting.

Hello,
I am an event technician and would like to use my Portapack to do RF coordination. 

For that, I need the ability to scan a certain frequency range and save the output in a .csv, .txt, or .spa format so I can import it in Shure Wireless Workbench (the control software of the wireless microphones).  

I tried doing it myself, but unfortunately, I lack the knowledge to write the code I need (ChatGPT and others aren't that great either...)

[](https://content-files.shure.com/Pubs/WWB/de-DE/de-DE/c_e28a4b1d-34e5-4b2a-86dc-db0277cb43d0.html)

gullradriel, 2025-01-21:
Already possible, at least in Recon.
It's having single frequencies and ranges scanning ability, with autosave on matching squelch level.
The output file is a freqman file, which is, by definition, a comma ',' separated CSV file.
See https://github.com/portapack-mayhem/mayhem-firmware/wiki/Recon.

gullradriel, 2025-01-21:
Workflow:
- prepare yourself a freqman file with your frequencies / ranges in it, and put it on the SDCARD in FREQMAN directory (see freqman file syntax [here](https://github.com/portapack-mayhem/mayhem-firmware/wiki/Freq-manager) )
- optionnal: prepare an empty output file and put it to the SDCARD
=> HINT<= use https://hackrf.app to quickly upload little files to the SDCARD without taking it out of the unit
- launch Recon
- click CONFIG
- select your prepared input file by clicking "select input file"
- select your prepared output file by clicking "select output file" or click on the underlying filename to manually type for a new name (may be RECON_RESULT by default)
- check autosave
- check continuous
- you may uncheck 'clear output file at start' if you wan to keep the result or clean the file manually
- click save
On main GUI:
- adjust the needed number of locks packets to match a frequency. The less nblocks, the more sensitive and prone to error. The more nblocks, the more precise but it will need a longer activity to record a frequency 
- adjust wait (W) and lock (L) timers. Wait is the time you are staying on a locked frequency, Lock is maximum time we spend on a frequency to lock it
- adjust the squelch level. That's the trigger level for a frequency to produce locks and match. It will be a thing to adjust yourself
=> HINT <= I'm usually having good results with the following defaults:
- nblocks:3 , W,L: 1000,100  (a wait of 1000 is a 1s wait on a locked frequency, switching audio and speaker on during the lock)
- nblocks:3 , W,L: 0,100  (a wait of 0 is a special setting that will not switch to audio and just record the match)

closed 2025-01-21 by gullradriel