portapack-mayhem/mayhem-firmware #2488: Unknown battery level and or charging status Portapack H4M 2.1.0

v-olmedo, 2025-01-21:
### Describe the bug.

I have let the portapack run and run out of battery.  Now, no matter how long I have it plugged in it doesn't charge or report the level of charging 

### Reproduction

1. Use until battery is drained.
2. Plug in.... for 24+hours
3. Battery reports 2% and no higher.
4. Overide battery calculations
5. enable screen time out.

### Expected behavior

Battery should charge

gullradriel, 2025-01-21:
Hmm.
Looks like you are not keeping the switch on the right on up position when charging. If it's in down position, the unit is powered up but not the charging unit nor the battery.

gullradriel, 2025-01-24:
Is your problem fixed ?

v-olmedo, 2025-01-24:
Yes!

closed 2025-01-24 by gullradriel