portapack-mayhem/mayhem-firmware #2489: receive doesn't work due to screen

DiamondRoPlayz, 2025-01-22:
### Describe the bug.

I'm just trying to listen to the fm radio through audio or sdr++ but it doesn't work because the screen cannot be turned off while on hackrf mode, receive only seems to work when i turn the display off but with hackrf mode it always stays on so it doesn't work.

### Reproduction

enter Audio or use sdr++ etc
tune to the radio frequency in your area of choice
just static even while outside of the house

### Expected behavior

Playa the radio audio

### Anything else?

audio seems to only work while screen is turned off but cannot do that through hackrf mode.

gullradriel, 2025-01-22:
It's not a bug. It's either you,your device, your câble or your computer.
There is no such thing as audio not working because the screen is on.

DiamondRoPlayz, 2025-01-22:
> It's not a bug. It's either you,your device, your cable or your computer. There is no such thing as audio not working because the screen is on.

not even related to the cable its the hackrf itself that audio is just static while the screen is on could be the firmware bugged

gullradriel, 2025-01-22:
> > It's not a bug. It's either you,your device, your cable or your computer. There is no such thing as audio not working because the screen is on.
> 
> not even related to the cable its the hackrf itself that audio is just static while the screen is on could be the firmware bugged

No man. While the firmware is not exempt from bugs, your case is clearly not a firmware bug.

Join discord and share a video of your 'bug'.

closed 2025-02-14 by zxkmm