portapack-mayhem/mayhem-firmware #2491: Mouse Jacking

Viarety, 2025-01-24:
### Description of the feature you're suggesting.

Is the Mouse Jacking feature feasible? Is there a way to add it as an application in the mayhem firmware?

zxkmm, 2025-01-25:
please provide more details of this.
I googled it, and found this site: https://www.mousejack.com/
In the site above, I click `Technical Details`, and it says "page not found"
So the code / hardware / technical details or even a simple "what it can do" isn't available.

zxkmm, 2025-01-25:
After searching, I found these:
https://web.archive.org/web/20160317040145/https://www.bastille.net/technical-details
https://web.archive.org/web/20160520212537/https://www.bastille.net/affected-devices

for whoever cares of it

Viarety, 2025-01-25:
Mouse Jacking is an attack on wireless mice connected to a computer via a USB radio channel. Under this attack, the mouse will start behaving uglily, spinning in all directions, pressing random keys, etc.

This attack has already been implemented on the flipper zero using the Nrf24 module, given that the HackRF One has an impressive radio range, I'm sure it can be implemented.

Viarety, 2025-01-25:
https://m.youtube.com/watch?v=gETzXNi3nO0

zxkmm, 2025-01-25:
Thank you very much! 
I think I can take a look of it's protocol and see how hard is it

Viarety, 2025-01-25:
I would really appreciate it if you could implement it.

zxkmm, 2025-01-25:
Thank you! I maybe not able to, maybe someone else can. Meanwhile I’ll take a look of it’s code.

Viarety, 2025-01-27:
I found out the frequency of interaction of wireless mice of this type: 2430 MHz (not all)

Viarety, 2025-01-30:
Please give me feedback as soon as you are ready, whether you have accepted this offer or not.

zxkmm, 2025-01-30:
I apologize but I haven’t read and investigate them yet.
And I didn’t touch baseband things much, I read and tried to understand some, but they are very hard. 
I can take a look but I think I’m mostly not capable to do that. But probably in next month or so. Or even better, that someone here do that before me

Viarety, 2025-01-30:
In general, if something works out / doesn't work out or there is any information, please leave feedback.

zxkmm, 2025-02-20:
I checked it's code when I got some free time in subway, the protocol is too complicated to me. 
I personally suggest if replay works then you just use replay.

zxkmm, 2025-02-20:
> whether you have accepted this offer or not

It's not quite real an "offer" since no bounty for it.

Viarety, 2025-02-20:
Sorry,google translator problems.

Viarety, 2025-02-20:
Anyway, I'll try to do it myself.

eried, 2025-06-23:
There is also the issue that the nrf24-related security issues have been fixed on all modern mices and keyboards, so unless there is newer vulnerabilities this would be very limited to a old subset of devices (I have an old keyboard just to test this 😂, I am not sure if it even worked... this was with a flipper)