portapack-mayhem/mayhem-firmware #2493: Ble RX, after 30 secs, try to save and have the "M0 Guru Meditation" Message: "System halted" ... Press DFU ...

NanoFile, 2025-01-25:
### Describe the bug.

Hi,
I've been starting  using the Ble Rx fonction and it look like after 10 founds MACs, if I try to save as CSV file, it crash.
If it is just 5 or 6, it's fine.
Here are pictures,

[image]

I click on "Export CSV":

[image]

Enter any name , then press "OK" :

[image]

Hope that help.
Richard.

### Reproduction

1. Select Ble Rx.
2. Wait for few received devices, more than 5 or 6.
3. Click on Save as CSV
4. Give a quick name.
5. Click on "OK".
6. Get the error ""M0 Guru Meditation" Message: "System halted" ... Press DFU ... "
7. Press "DFU" give the Stack Dump on the screen.
8. Press Reset button to reset device.

### Expected behavior

1. Should save .CSV file to SD Card and return to main screen.

### Environment/versions

Firmware version 2.1.0

### Anything else?

This looks promessing to review Bluetooth device around.

zxkmm, 2025-01-27:
However I noticed you are using freq shift (ham it up). 
I think it’s not related to this issue, but also it doesn’t make sense to use it for BLE. BLE adverse is in specific 3 frequencies and it’s meaningless to shifting them.

gullradriel, 2025-01-27:
I can reproduce the bug when the list of entries is becoming bigger than the screen. 
Maybe there is a bad copy of the list instead of a reference to it when saving it, or the save functions are too hungry.
I'll try to have a look at it, tonight or tomorrow.

NanoFile, 2025-01-27:
> However I noticed you are using freq shift (ham it up). I think it’s not related to this issue, but still it doesn’t make sense to use it for BLE. BLE adverse is in specific 3 frequencies and it’s meaningless to shifting them.

Hi,
yes your right about the "Ham it Up" was activated, I just remove it and redo my tests and as soon as the screen is filled, when you try to export the CSV file, it still have the same error.
Thank you for testing this :)
Richard.