portapack-mayhem/mayhem-firmware #2504: button remapping

null-p4n, 2025-01-28:
### Description of the feature you're suggesting.

Hi there,

Recently, due to user error (me being an idiot), my down button is missing. I tried to repair it, but either I screwed something up or the H2 board is so cheap that when I removed the button to replace it, the trace wires came off too. So now I’m stuck with one missing button, and I can’t scroll down.

Given that the PortaPack has just four buttons and one rolling dial, there's no unused button to remap. However, implementing a button remapping feature could allow users to assign essential functions to different buttons or combinations, potentially mitigating issues like this. For example, combining the rolling dial with another button could simulate the "down" function, providing a workaround for hardware failures.

zxkmm, 2025-01-29:
Personally I think it's too edge case that it's not worth to implement (if re-map, then every button press cost more resources, and also to implement that, it cost more rom spaces.

For example if your iPhone's upper part screen is broken, it doesn't make sense to ask apple to make a "only half screen mode"

zxkmm, 2025-01-29:
BUT it's easy to make your own firmware. But I think this edge case won't go to official mayhem settings

null-p4n, 2025-01-29:
> Personally I think it's too edge case that it's not worth to implement (if re-map, then every button press cost more resources, and also to implement that, it cost more rom spaces.
>
> For example if [...]

This isn't about demanding a full UI overhaul because a user made a mistake. I fully acknowledge my own error, and the point is to help others avoid a simple design flaw that could lead to hardware issues. A remapping solution was a reasonable way to mitigate that.

While "just fork your own and do it" is a common response, it's neither a real solution nor helpful advice. The discussion here is about practical improvements, not dismissing concerns as edge cases.

gullradriel, 2025-01-29:
Thank you for your suggestion! At the moment, we don't plan to make this modification ourselves, as it doesn't align with our current priorities. However, if you’d like to implement it, we’d be happy to review a pull request.
Let us know if you need any guidance on how to proceed!

closed 2025-01-29 by gullradriel