portapack-mayhem/mayhem-firmware #2505: Pulse Signal Generator (Signal Generator new feature)

voxo22, 2025-01-31:
### Description of the feature you're suggesting.

It would be nice to have pulse signal generator, which is definitively missing among HackRF apps. It could be integrated into existing Signal Generator app. The generator would produce only pulsed CW signal (can be integrated into CW selection), with custom settings:
- pulse time (e.g. 5 µs to 10s) - Don't know, how short the pulse is possible, the shorter, the better. Also, the more detailed setting, the better (e.g. format 0.00000000s) like frequency selector.
- pulse interval (the same as pulse time) - pause between pulses
- loop / single pulse (e.g. on START button)
All other settings (freq, tx power, etc,) the same as in signal generator app.
Thanks!

daviderud, 2025-03-01:
#2537 can be a starting point. Pulse duration now is = (1/tone freq)*0.25 

Possible next steps: Increase resolution of tone frequency down to 0.000xx Hz; make duty cycle selectable possibly with high resolution.

voxo22, 2025-05-15:
@daviderud exactly this code can be used as pulse generator feature: https://github.com/jfdelnero/rf-tools/tree/master/src/rf_jammer
Some parameters are not mandatory to keep, but jam_interval (pause between pulses), jam_duration (pulse length) and sample_rate are very important. This little piece of code can generate impulse as short as <100 ns while using 10 MHz sample rate.