portapack-mayhem/mayhem-firmware #2510: Add 978 mhz option to ADSB-receive | option to hop between 978 and 1090 at 1 or 2 second interval.

chris451gh, 2025-02-06:
### Description of the feature you're suggesting.

ADSB out transmits on 1090 and 978 was added for broadcasts from ground to air and as a secondary ADSB broadcast frequency.
There is consistant activity there. Add a 978 mhz option so the ADSB  signals there can be received. Current app assumes only 1090 is needed. 
Some aircraft may only operate on 978. Also it may be possible to toggle between 978 and 1090 every second and still pick up traffic on both frequencies. Additional data not in the ADSB format may be transmitted on 978 by an airport with ASR-11 secondary radar.
It may also broadcast mode s locations to aircraft so they see it on ADSB. A mode where the tuner goes to 978 for 1 second then 1090 for the other second may be an effective way to get both signals at nearly the same time. To determine if there is a signal present tune to AM 978 and you should pick up 978 pulses 1 per second. Some transmissions may take 2 cycles (2 seconds) to fill in all the data so maybe a 2 second hop would be more effective. For a test as a minimum just add ability for tuning to 978 manually At the moment there are 2 ups flying within range at least one of them is on 978.. On 1090 there are interrogated responses and on 978 they just broadcast whenever enabled.

chris451gh, 2025-02-08:
So I tuned a Hackrf running SDRAngel to 978 and I see packets going by about 1mhz bandwidth and much longer than the pulses on 1090. It is a different format, and github has a repo to decode it here:  https://github.com/zaitcev/ruat

They call the transponders UAT but since they still require 1090 its cheaper to upgrade the transponder then to add another box that has to be integrated with the GPS.  
So hopping may not be practical it needs a different decoder. There are signals there, and they do not decode as adsb format.

gullradriel, 2025-02-15:
Interesting. I don't know when one will be able to add the decoder, but it's definitely interesting.