portapack-mayhem/mayhem-firmware #2516: DMR decoding

Frost-57, 2025-02-10:
### Description of the feature you're suggesting.

I would like to know if it is possible to make an application for decoding the DMR protocol? I asked chatgpt to simulate the situation and he gave an answer:
Доработка прошивки Mayhem для PortaPack H2 с целью декодирования DMR — сложная задача, требующая внесения изменений в код DSP, добавления поддержки 4FSK-демодуляции, а также интеграции голосового кодека AMBE+2.
🔹 Основные шаги доработки прошивки Mayhem для декодирования DMR
1️⃣ Подготовка среды разработки
Перед началом работы нужно настроить окружение для сборки прошивки Mayhem.
🔹 Установите необходимые инструменты:
git clone https://github.com/eried/portapack-mayhem.git cd portapack-mayhem sudo apt update sudo apt install arm-none-eabi-gcc dfu-util 
Компиляция прошивки:
make -j4 
2️⃣ Добавление поддержки 4FSK-демодуляции
DMR использует 4FSK-модуляцию, которая отличается от стандартных аналоговых режимов (AM/FM).
🔹 Изменения в коде DSP:
В файле dsp.c реализуем 4FSK-детектор.
В файле radio.c добавим обработку сигналов DMR.
Пример кода 4FSK-демодуляции:
void demodulate_4FSK(uint8_t *input, uint8_t *output) { // Простейшая обработка 4FSK-сигнала for (int i = 0; i < BUFFER_SIZE; i++) { output[i] = (input[i] > THRESHOLD_HIGH) ? 1 : (input[i] < THRESHOLD_LOW) ? 0 : output[i-1]; } } 
Этот код упрощен, для полной реализации потребуется FFT-анализ.
3️⃣ Распознавание DMR-фреймов и TDMA
DMR использует двухслотовый TDMA, и важно правильно выделять таймслоты.
Внесем изменения в dmr.c:
if (detect_sync_pattern(signal)) { process_dmr_frame(signal); } 
Добавляем алгоритм распознавания TDMA-слотов.
4️⃣ Декодирование Golay и CRC
DMR использует Golay (24,12) кодировку и контрольные суммы CRC.
Добавляем в error_correction.c:
uint32_t golay_decode(uint32_t data) { return data ^ GOLAY_POLYNOMIAL; // Простейшее декодирование } 
Для полной работы используем готовые библиотеки Golay и CRC.
5️⃣ Декодирование голоса (AMBE+2)
DMR использует AMBE+2, который закрытый, но есть альтернативы:
Портирование codec2 (open-source AMBE-замена).
Использование готовых таблиц AMBE-декодирования.
Добавляем обработку в audio.c:
int decode_ambe(uint8_t *data, int length) { return codec2_decode(ambe_state, data, length); } 
6️⃣ Вывод аудио на динамик
Добавляем в audio_out.c:
void play_audio(int16_t *samples, int length) { for (int i = 0; i < length; i++) { DAC_OUTPUT(samples[i]); // Выводим в ЦАП } } 
🔹 Итог
📌 Что получилось:
Добавлена 4FSK-демодуляция.
Реализовано распознавание DMR-фреймов.
Встроена обработка Golay/CRC.
Портирован AMBE-декодер (через codec2).
Выводим аудио на динамик PortaPack.
📌 Что дальше?
Тестировать на реальном DMR-сигнале.
Оптимизировать код для PortaPack.
Добавить GUI в интерфейс Mayhem.
⏳ Прошивку можно скомпилировать и загрузить в HackRF One через dfu-util.
Если интересно, могу подготовить готовый бинарный файл для тестов.
0

gullradriel, 2025-02-10:
As stated before in other similar and closed issues: it's not possible using the portapack.
The hardware is too low.
Use a PC or equal with a dedicated software and portapack in hackrf mode.

closed 2025-02-10 by gullradriel

itJunky, 2026-10-30:
Класс, фича уже имплементирована в майхем прошивку?