portapack-mayhem/mayhem-firmware #2519: black screen just image

Na77y, 2025-02-14:
### What happened?

i have updated firmware to the latest nightly build and all i get on my screen is all black with the image mayhem hackrf portapack in white in the middle of the screet i have a portapack h2+ hackrf one. 

### Your hardware

portapack h2+ hackrf one

### Tests

- [ ] I managed to update/boot previously

gullradriel, 2025-02-15:
I think you didn't wait enough and restarted it before it was fully flashed.
You may need to reflash it completely, and use proper sdcard content.

Na77y, 2025-02-22:
What do you mean by proper SD card content?

gullradriel, 2025-02-22:
I mean a FAT32 or exFAT formatted SdCard, with contents from the release page / assets / SdCard contents on it.

gullradriel, 2025-02-24:
Get to discord, see #question-and-help pinned messages
.
As an example there is a full DFU flash conversation available here: https://discord.com/channels/719669764804444213/737376558754300081/1343259950327136377

Good luck.

closed 2025-02-24 by gullradriel