portapack-mayhem/mayhem-firmware #2521: FT8/ WSJTX decoder and transmit.

FixifyAI, 2025-02-16:
### Description of the feature you're suggesting.

Decoding (and maybe transmitting) FT8 and other digital modes would be nice.

gullradriel, 2025-02-16:
FT8 is unrealistic with the hackrf/portapack hardware. Do it with hackrf or portapack in hackrf mode connected to a PC. 

WSJTX is a decoder. Use it on your PC with hackrf or portapack in hackrf mode.

SSTV: duplicate of https://github.com/portapack-mayhem/mayhem-firmware/issues/2045

closed 2025-02-16 by gullradriel

metawops, 2025-10-15:
Unrealistic exactly why? 😳

PeterGrace, 2025-11-01:
> Unrealistic exactly why? 😳

@metawops see https://github.com/portapack-mayhem/mayhem-firmware/issues/2316#issuecomment-2994124052 -- good thorough explanation about why FT8 can't happen, sadly.