portapack-mayhem/mayhem-firmware #2525: What is the difference between portapack-mayhem-firmware.bin and portapack-mayhem_OCI.ppfw.tar?

henryclw, 2025-02-18:
### What happened?

I'm using this method to update the firmware `Copy firmware all in one tar package into SD card, and offline update with Flash Utility app in portapack`
However there are two different file that I could select, `portapack-mayhem_v2.1.0.bin` and `mayhem_v2.1.0_OCI.ppfw.tar`. Previously I always update using the bin file. May I ask if there is any difference between these two files? I tried to search in the wiki but could find any thing. (Maybe I just missed it)

### Tests

- [ ] I managed to update/boot previously

gullradriel, 2025-02-18:
.bin => only flash content
.ppfw.tar => both flash and external sd card apps

You should use the ppfw.tar file if you want to update both firmware and external sdcard apps in one pass.

closed 2025-02-18 by gullradriel

henryclw, 2025-02-19:
Thank you for your kind reply. It's very helpful.

I just updated the documents like this: https://github.com/portapack-mayhem/mayhem-firmware/wiki/Update-firmware/27ed828661058680de07d3d36f762efd2e0f9d31
Editing the method3, telling user what does these files mean. I hope this would be helpful.

gullradriel, 2025-02-19:
Nice. Good edit, we are keeping it.