portapack-mayhem/mayhem-firmware #2527: Seems to be bricked after opening

ukarLala, 2025-02-21:
### What happened?

I'll go over what happened step by step. Excuse any glaring ignorant mistakes please, I have worked 70 hours this week and figured I'd just have a quick look inside this thing in the few minutes I had spare before bed.

I wanted to mount a small speaker into the case of my H2+. Been working great but obviously without sound there's limitations. When it arrived it came with a small red/black cable and connector for the J9 connection inside the Portapack case. I, seemingly wrongfully, assumed this was an audio connection.

I opened the case, it was a bit of a pain in the but and 3 of the 5 buttons few off in all directions. I detached the Portapack board from the HackRF. I carefully attached the J9 cable with nothing on the other end and the. Reattached the Portapack board to the Hack RF and attempted to power on, planning to navigate to the audio listener and touch my cables to a small speaker input and see if I got any sound out.
The screen came on but just showed white. I pressed my OK button and it briefly showed the start up page and then went black. I powered it on again and again saw the Start up icons briefly and then black screen. I tried again and got nothing.

I plugged into my usual power input and got the same thing again but after the screen went black the bottom right blue light around the rotary knob stayed lit whilst the other three flashed. Every 6 or so flashes the RF 1v8 and the 3v3 LEDs would flash once.

I tried powering whilst changing display driver and nothing changed. At this point the display on my power supply read "Circuit Protection Active! Please stop using power supply"
I unplugged my USB and the power supply went back to usual 

Question? What has happened? I feel like something somewhere shorted and cause some insane battery drain.
I disconnected from the J9 port and tried to power up again but got nothing.

This was rather expensive for it to be killable just by removing the case. I have built multiple PCs and regularly work on electronics. I was dismantling this at the work station where I work on appliance parts and control boards. I was definitely tired when opening this device but short of making assumptions about connector J9 I don't believe I made any other mistakes here (sure I could have disconnected the battery before plugging into the port but that seemed like overkill)

I was running stable 2.1.0

### Tests

- [x] I managed to update/boot previously

zxkmm, 2025-02-24:
I apologize in advance. Forgive me but I can’t really pick valid info from the long text filled with background/emotion/monologue/useless expressing. And I tried my best to read carefully but it seems some necessary information still missing from your text.

Can you please explain what you exactly did? Like this: 1. Connect x to y. 2. Unplug x. 3. x not working anymore. Etc.

And here from your description I noticed several things:

> I, seemingly wrongfully, assumed this was an audio connection.

And then? What did you do?

> I carefully attached the J9 cable with nothing on the other end and the. Reattached the Portapack board to the Hack RF and attempted to power on, planning to navigate to the audio listener and touch my cables to a small speaker input and see if I got any sound out.

Don’t do this, useless and dangerous approach.

> This was rather expensive for it to be killable just by removing the case. 

Skill issue.

zxkmm, 2025-02-24:
So overall you may did these wrong:
- lost p.mem and make screen boot option wrong, need to follow the “won’t boot” guide.
- you fried something when doing touch to test with wires. Because it’s just simply dangerous. They can be shorten circuits when you didn’t paying attention, or it jump too fast and killed some chips.
- you plugged battery into speaker port (or reverse) and killed something.

ToxxikHalo, 2025-02-24:
Yea as bro above said, you gotta explain EXACTLY what happened minus the emotional melodrama, we understand it sucks...
Have you tried reflashing? Which method? 
You plugged in the audio leads to the audio jst connector and left them open right? Then tried to manually contact them to a speaker?  Could they have accidentally touched an part of the board while u were maneuvering and manipulating the leads??? I'd guess that the most likely culprit...

ukarLala, 2025-02-24:
> Yea as bro above said, you gotta explain EXACTLY what happened minus the emotional melodrama, we understand it sucks... Have you tried reflashing? Which method? You plugged in the audio leads to the [...]

I never reached the stage of trying to manually connect the open ends to a speaker. At all times these speaker wires hung out the side of the device carefully contacting nothing but air.

ukarLala, 2025-02-24:
> I apologize in advance. Forgive me but I can’t really pick valid info from the long text filled with background/emotion/monologue/useless expressing. And I tried my best to read carefully but it [...]

I wrote my original message to show

1) Self deprecation and humility
2) Explanation of my goals
3) My Plan
4) My Method
5) The end result

I do not apologize that this was not to your liking.

ToxxikHalo, 2025-02-24:
Relax bud, nobody was trying to talk down on you, I get your upset and its
cool we just needed to clarify the details to try and figure out a solution
that's all...
It's odd to me that it turned on at first then blacked out... makes sense
if something shorted maybe... look at the board again when you can and look
for any obvious damaged components...

zxkmm, 2025-02-24:
@ukarLala sorry, I’m not sure if you aware that we are not selling anything nor profiting from this project. 
The help from @ToxxikHalo and I also is voluntary.

You may however complain about bad quality to the seller/manufacturer, but we can’t help you about quality issue even if you mentioned that you build computer or other tec stuff often.

zxkmm, 2025-02-24:
Also I provided solutions for you to read, you ignored and you just put your bad mood here again. 

You’ve already lost my further help. And I’ve just blocked you from my account. Even this acc is probably an alt of someone else.
 
If you wanna futures help, I recommend you to write a clean step of what you did below, instead a chunk of unformatted text mixed with not related information and still lack of details, and wait others to help you, if they are willing to.

gullradriel, 2025-02-24:
The user likely shorted something on their HackRF Portapack while trying to connect a speaker to the J9 connector, which they mistakenly assumed was for audio.
After reassembling the device, the screen initially showed a white display, then briefly flashed the startup icons before going black. Now, the device does not power on properly, and their power supply reports a "Circuit Protection Active" warning, indicating a possible short circuit or excessive current draw. 
The flashing LED pattern suggests a failure in power regulation, potentially due to damage to the Portapack board, the HackRF, or both.

Contact your reseller, go to discord to seek hardware help.

closed 2025-02-24 by gullradriel

ukarLala, 2025-02-24:
> Relax bud, nobody was trying to talk down on you, I get your upset and its
> cool we just needed to clarify the details to try and figure out a solution
> that's all...
> It's odd to me that it turned on [...]

Thank you for the human responses and the advice. I appreciate the time you have taken dude. Have a good one

gullradriel, 2025-02-24:
> I did not complain about bad quality and I read the steps to take given but felt it important to correct the errors of the responses. Such as J9 not being the speaker connector... It is Such as if I disconnected the battery... I did not
> 
> If you don't want to read a formatted block of text written like a human you don't have to. You can just move right along

The issues are mainly for firmware related problems / enhancements, which is not your case.

Please seek help with the reseller and or on discord. Maybe some more people can help you there, as there are more hardware guys online on discord than on the issue on github.

Also there was a documentation for proper speaker installation, it's here: https://github.com/portapack-mayhem/mayhem-firmware/wiki/Internal-speaker

I will not comment that issue anymore. 

Thanks for your comprehension.