portapack-mayhem/mayhem-firmware #2530: APRS RX & TX

FixifyAI, 2025-02-26:
### Description of the feature you're suggesting.

It would be nice to send and receive APRS all in the same app, instead of switching back and forth. If this is already possible please let me know.

gullradriel, 2025-02-26:
HackRF is half duplex. So not possible to send and receive at the same time.

f1ghy, 2025-02-26:
Hi, It is not a mater of tx and rx in the same time, but a mater of receive, and later tx, as "microphone" app does. As usual talky walky. APRS does not need full duplex.
As well it could be fine to use GPS positionning. 
Thanks ;o)

strijar, 2025-04-10:
I agree - we need to combine them in one app. I can take it upon myself. Especially since I plan to create a separate TRX group for such apps.