portapack-mayhem/mayhem-firmware #2532: rolling code reciever

scpitt, 2025-02-27:
### Description of the feature you're suggesting.

code needs to transmit a jamming freq next to the target freq and allow for recording whilst transmitting if possible ? can you use record and tx simutaneously ?

zxkmm, 2025-02-27:
it's called **rolljam**

However limitations exists:
- you need two transmit sources (one jamming and another transmit the *last* code you captured) and one receive source (to capture the real fob's all signal).
- this attack is barely working, with manu limitations.

Thus personally i think it's not realistic to make this happen. 
But there are chance that can use 2 or 3 hackrfs and manage to sync them in some way, if anyone interested to implement this, feel free to re-open.

zxkmm, 2025-02-27:
> can you use record and tx simutaneously ?

it's not about this thing, but for this specific question: no. hackrf is half duplex.