portapack-mayhem/mayhem-firmware #2545: Overdischarge warning / autoshutdown

auroJimen, 2025-03-06:
### Description of the feature you're suggesting.

I noticed that on the H4M it is quite easy to accidentally overdischarge the battery. 

Ideally I think the system should autoshutdown in that situation (I don’t get why the battery managing circuit doesn’t do anything to prevent it) but maybe it could be implemented in software, or at least show a pop-up to let the user know if they haven’t noticed.

gullradriel, 2025-03-09:
I don't think we can shutdown the unit from the firmware.

zxkmm, 2025-03-10:
As gull mentioned, it’s not possible for this mcu to shut down itself.

For over discharging I think it’s tricky, imagine that when you are asleep and the device started beeping. 

I think it’s a good habit to turn off when you don’t use it, and set screen timeout also helps. Since this device boots very fast, you can completely use the switch the same way as the “lock” button of phones.

auroJimen, 2025-03-12:
I was thinking more along the lines of showing a "danger" version of the battery icon or something similar when voltage is in overdischarge levels. But i do agree it is not necessary if one's carefull with the device. Thanks for considering my idea, i agree with closing the issue.

zxkmm, 2025-03-12:
Thank you for the clarification. Previously, I misunderstood your request.

So you meant that we can show a message when the battery is too low (like under 5%)?
I think this is a good idea, but I'm not sure if it's necessary, because that the battery itself have protection board to prevent it over discharging itself (over discharging that can actually cause damage). So it's usually probably won't over discharging itself, unless some revision got bad quality battery or bad design?

And for another "over discharging" from my understand is that someone just simply forgot to turn off and when they wanna use it, they found it's dead. In this case I think it could be good, but still if they forgot to turn off, maybe they won't notice the warning either, or we can use speaker to beep, but a bit scary?

Anyway, I will re-open this for future discussion

auroJimen, 2025-03-12:
Thanks for answering so fast!

I've been using my h4m mayhem from opensdrlabs for a week and i've seen it dip bellow dangerous voltage levels for the lipo several times. I'm not sure if the voltage is being wrongly reported or if maybe after a few cycles the overdischarge protection of the battery ic will start working. 

Looking around online i did see a couple of people reporting the same problem in the h4m. Thats why I opened the issue. One person said they talked to opensdrlabs support to no avail hence my idea to issue a warning in case more h4m users are letting their deviced go bellow 2.5V without noticing.

zxkmm, 2025-03-12:
lemme test mine XD. Maybe they are using bad quality battery?

htotoo, 2025-03-14:
Iirc, there is an over discharge protection in the battery. Maybe around 2.75V?

gullradriel, 2025-06-22:
Not a Mayhem firmware problem. Open an issue or a question/dispute with the reseller. 
As stated we can not shutdown the unit from the firmware.
There is nothing we can do.