portapack-mayhem/mayhem-firmware #2546: Pocsag did not recognize the bitrate right

fkb83, 2025-03-06:
### Describe the bug.

Pocsag thinks it´s a 512 bit transmission, but it´s a 1200 bit one

### Reproduction

Switch on, goto Pocsag app

### Expected behavior

Finest Decoding

### Anything else?

We have 3 main transmitter for pocsag in germany. the have some sort of preamble which let Pocsag think it´s a 512 bit transmission. After that preamble pocsag sticks onto that speedrate. But the transmission comes with 1200. This leads to complete undecoded messages, all violet. 
If i switch right after the preamble to that frequency the speedrate will be recognized right and decoding works fine.

gullradriel, 2025-03-09:
Which firmware version ?

fkb83, 2025-03-09:
Tested with 2.1.0 and the actual nightly. Behaviour is the same

janphkunzgit, 2025-03-11:
Same here. Maybe a quick fix would be the possibiltity to manually select the baud rate in addition to the automatic detection?

gullradriel, 2025-04-08:
> Still no Updates 🤔

No, but you can get your hands on it maybe ?

f1ghy, 2025-04-09:
Hi, Time ago, you had to choose 1200 or 2400 Baud in Config page. I regret this time too. An easy way could be to come back to this manuel choice, why not "auto/1200/2400".
Thanks you.
;o)

bloudman648, 2025-05-03:
hmm..

bloudman648, 2025-05-05:
I also did that, i heard older build versions does have the option

gullradriel, 2025-05-05:
> I also did that, found nothing. i heard older build versions does have the option?

Yeah, it was changed to an automatic detection.
We should add as @f1ghy said a switch with an auto/1200/2400 setting.

MeshMouse, 2025-08-24:
I'm having very similar issues with decoding POCSAG. Sometimes the packets are decoded perfectly, but most of the time either nothing is decoded or I get garbled decodes.

htotoo, 2025-09-14:
Join discord, and check the #test-drive chan

htotoo, 2025-11-18:
I implemented a manual baud rate contfol in the pocsag app's settings menu. you can now select auto, 512, 1200, 2400.

test fw: https://discord.com/channels/719669764804444213/722101917135798312/1440302734409601105