portapack-mayhem/mayhem-firmware #2547: SubGHz Rolling Code Protocol Support

Axolotl5903, 2025-03-07:
### Description of the feature you're suggesting.

Hi Mayhem Developers!

I hope you're doing well!

I wanted to suggest adding SubGHz Rolling Code Protocol support to the Mayhem Firmware. I understand that implementing this feature—and ensuring it functions correctly—could be challenging. However, I was wondering if it might be possible (or easier) to port the existing rolling code signal decoding and emulation features from the Flipper Zero firmware, since both the Mayhem and Flipper Zero firmwares are written in C.

If porting these features is feasible, I believe it would be an incredible addition to the Mayhem Firmware, greatly expanding its capabilities. I'd love to hear your thoughts on whether this is something that could be considered.

Thanks for your time, and I appreciate all the work you do on the firmware!

Best regards,
Chris

htotoo, 2025-09-09:
It is doable, and the current code is based on Flipper's code, so wouldn't be so much difference, but the FW space limits it.

limbo111111, 2025-09-29:
Is it possible to run These memory-intensive tasks on the mdk module? 
Greets! keep Up the good Work!