portapack-mayhem/mayhem-firmware #2548: è possibile lingua italiana nel menu schermata portapack h2 ?

FAUSTO2163, 2025-03-07:
### Description of the feature you're suggesting.

sarebbe per me molto utile il menu schermata portapack h2 in lingua italiana , spero tanto che esista un file da inserire per ottenerla, grazie

[Schermate ADD-ON mio.docx](https://github.com/user-attachments/files/19128704/Schermate.ADD-ON.mio.docx)

NotherNgineer, 2025-03-08:
Supporting multiple languages would require some effort since we have such a small amount of flash memory, and text strings are currently spread across every application module.  We'd also have to support some additional font characters (e.g. in the small font size).  Implementation wise, to save memory we could perhaps make language a compile option and release a separate firmware binary image for each language (text strings could be moved to header files that include every string in multiple languages).  Which languages should we support?  French, Spanish, Italian, German, which others?

FAUSTO2163, 2025-03-10:
> ATTENZIONE: questa e-mail proviene da una fonte esterna. Presta sempre attenzione ai link e agli allegati, anche quando il mittente sembra, apparentemente, essere interno.

Buongiorno, sarebbe molto utile in lingua italiana , ma se avete problemi va bene anche nella lungua inglese, salve Buona Giornata.

FAUSTO FUMAGALLI

Banca di Credito Cooperativo di Milano S.C.

Filiale: 27 INZAGO (MI)

Via Cavour n° 8 cap 20065 Inzago (MI)

Email: ***@***.***

Tel. 02/953161
Fax. 02/9274297

[cid:86bdfdb0-e0f8-47b2-8bcc-d0fed4c2d338]

La presente comunicazione e ogni suo allegato è destinata esclusivamente al soggetto indicato quale destinatario o ad eventuali altri soggetti autorizzati a riceverla. Essa contiene informazioni strettamente confidenziali e riservate, la cui comunicazione o diffusione a terzi è proibita, salvo che non sia stata espressamente autorizzata. Se avete ricevuto questa comunicazione per errore, Vi preghiamo di darne immediata comunicazione al mittente e di distruggere il contenuto. La informiamo inoltre che l’utilizzo non autorizzato del messaggio o dei suoi allegati potrebbe costituire reato. Banca di Credito Cooperativo di Milano S.C. tratta i dati personali in linea con il Regolamento Europeo 2016/679 (GDPR). Per informazioni può consultarsi la sezione “Privacy” sul sito della società.

htotoo, 2025-03-10:
But the real problem would be, supporting translations for all new apps, ..

closed 2025-10-16 by htotoo