portapack-mayhem/mayhem-firmware #2560: Signal number in OOKBrute

Viarety, 2025-03-11:
### Describe the bug.

Hello dear mayhem developers. I recently ran into this problem. In search of a signal for my gate, I used the OOKBrute function and was pleasantly surprised that the gate was open in just 5 minutes. After trying a few more times, I made sure that the 3040 signal in came12 exactly matches my gate. But when I tried to open them the next day, it took me a long time to scroll to the 3040 signal with the wheel, and I was afraid not to kill the wheel on my h4m. I think it is necessary to add a numeric input for the serial number of the signal so that such problems do not arise. It would be an even cooler improvement if these signals could be uploaded to convenient quick access or simply saved as .OPEN the file and use it fully after selecting the signal. Thank you for your time.

### Reproduction

when scrolling through the sequence number of the OOKBrute signal, only wheel movement is available, and there is no numeric input.

### Expected behavior

.

### Environment/versions

<Nightly 10.03.25

zxkmm, 2025-03-11:
I'm thinking of this too. 
It's easy to have this single app to have input support, but I'm thinking that make the numberfield widget has input support, which is tricky, because current prompt_input (or similar name I forgot) need you to have a living var at the full lifetime that enter until exit the input scene. 
So it's needed to re-design that function, which is hard, because it's not copying is the reason that notepad can handle big files in this device.

If it's necessary, we probably can just make this specific app can input, which is easy but not elegant.
CMIIW

Viarety, 2025-03-11:
I would be very grateful if you could do it at least somehow.

zxkmm, 2025-03-11:
Will do with the simple way today

zxkmm, 2025-03-11:
(If my parents didn’t return early and found out I’m not preparing for job exam instead of using computer.)

zxkmm, 2025-03-11:
https://github.com/portapack-mayhem/mayhem-firmware/pull/2561

closed 2025-03-11 by zxkmm