portapack-mayhem/mayhem-firmware #2563: How to flash firmware if make pcb, buy new blank CPLD to solder portapack h4m myself

zhouxule, 2025-03-13:
### What happened?

If I use the h4m gerber to make pcb, buy new blank EPM240T100C5N and solder the components myself, how to flash the firmware or which guide should I refer to? Can I just run the bat file or I need a programmer like usb blaster to flash the firmware for EPM240T100C5N? Thanks and best regards.

### Your hardware

hackrf, portapack h4m made by myself

### Tests

- [ ] I managed to update/boot previously

bernd-herzog, 2025-03-19:
It is possible to program the portapack CPLD from the usb serial console with the cpld_write command.

See https://github.com/portapack-mayhem/mayhem-firmware/wiki/usb-serial-console for more information.

RoyTse, 2025-03-27:
> ### What happened?
> If I use the h4m gerber to make pcb, buy new blank EPM240T100C5N and solder the components myself, how to flash the firmware or which guide should I refer to? Can I just run the [...]

你PCB板做了吗？我也想弄一个

zxkmm, 2025-04-06:
BOM not provided from OSSL. So currently it's not possible to make your own.

closed 2025-06-22 by gullradriel

zxkmm, 2025-06-22:
To those who read through, the PCB is fully open sourced now. Including BOM. Marked my previously comment as outdated.