portapack-mayhem/mayhem-firmware #2564: Add ability to click on a frequency in all spectrum views, to move the marker

BuddyPage, 2025-03-13:
### Description of the feature you're suggesting.

A feature that will make spectrum views far more effective and efficient. Instead of being required to use the knob to move the marker, make all  spectrum views clickable so that touching a frequency (or any point) with a stylus jumps the marker to that point. Not sure if this has already been suggested.

htotoo, 2025-03-14:
The marker selector click would be good!
But the freq change could cause problems i think, because of the accidental touches.

kniknoo, 2025-03-18:
I agree with OP, this is one of my biggest QoL requests right now.
To prevent the accidents, would it be possible to use the center button as a modifier key that enables tap or perhaps a long press that enables a single tap?

zxkmm, 2025-04-12:
I implement this.
long press: the ghost touch and bouncing of this screen is too bad to implement long press detect of touchscreen
touch to set cursor, click ok to enter: that's what I did.

gullradriel, 2025-04-14:
Implented in https://github.com/portapack-mayhem/mayhem-firmware/pull/2624 by @zxkmm

closed 2025-04-14 by gullradriel